ros/geometry2 #84: undefined reference to `tf2::BufferCore::_allFramesAsDot() const'

bricerebsamen, 2015-03-11:
after #74 I get this error
/opt/ros/indigo/lib/libtf.so: undefined reference to `tf2::BufferCore::_allFramesAsDot() const'

to reproduce: create a workspace with the ros/velodyne repo and the ros/geometry_experimental repo, both of them checked out to the latest version.

I bisected on ros/geometry_experimental and I confirmed #74 is the culprit

tfoote, 2015-03-11:
Is your workspace in /opt/ros/indigo? It's odd that you're getting that issue from /opt/ros/

I've seen the same issue one our new test farm: http://54.183.26.131:8080/job/Idev__geometry_experimental__ubuntu_trusty_amd64/6/console

But the old farm is passing: http://jenkins.ros.org/job/devel-indigo-geometry_experimental/  And I cannot reproduce this on my desktop.

bricerebsamen, 2015-03-11:
no I created a dedicated workspace in my home dir that contains only those 2 repos. So tf is pulled from `/opt/ros/indigo`
Note: I built using `catkin_tools`

tfoote, 2015-03-12:
You shouldn't be using symbols from /opt/ros that depend on a source overlay. 

However I've also attempted a rework to overload instead of have a default method. Can you try the fix_84 branch?

tfoote, 2015-03-12:
Hmm, I just noticed you're having issues with allFramesAsDot that's different than the allFramesAsYAML which i'm reproducing. 

Some more info: 
First pass is producing a void argument symbol, rming devel and build produces double
```
buildfarm@7b33dfc8942c:/tmp/catkin_workspace$ nm /tmp/catkin_workspace/devel_isolated/tf2_ros/lib/libtf2_ros.so | grep YAML
                 U _ZNK3tf210BufferCore15allFramesAsYAMLEd
buildfarm@7b33dfc8942c:/tmp/catkin_workspace$ nm /tmp/libtf2_ros.so | grep YAML
                 U _ZNK3tf210BufferCore15allFramesAsYAMLEv
```

buffer server linking successfully: 
```
Linking CXX executable /tmp/catkin_workspace/devel_isolated/tf2_ros/lib/tf2_ros/buffer_server
/usr/bin/cmake -E cmake_link_script CMakeFiles/tf2_ros_buffer_server.dir/link.txt --verbose=1
/usr/bin/c++       CMakeFiles/tf2_ros_buffer_server.dir/src/buffer_server_main.cpp.o  -o /tmp/catkin_workspace/devel_isolated/tf2_ros/lib/tf2_ros/buffer_server -rdynamic /tmp/catkin_workspace/devel_isolated/tf2_ros/lib/libtf2_ros.so /opt/ros/indigo/lib/libactionlib.so /opt/ros/indigo/lib/libmessage_filters.so /opt/ros/indigo/lib/libroscpp.so -lboost_signals -lboost_filesystem /opt/ros/indigo/lib/librosconsole.so /opt/ros/indigo/lib/librosconsole_log4cxx.so /opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex /opt/ros/indigo/lib/libxmlrpcpp.so /tmp/catkin_workspace/devel_isolated/tf2/lib/libtf2.so /opt/ros/indigo/lib/libroscpp_serialization.so /opt/ros/indigo/lib/librostime.so -lboost_date_time /opt/ros/indigo/lib/libcpp_common.so -lboost_system -lboost_thread -lpthread -lconsole_bridge -lboost_thread -lpthread -lconsole_bridge -Wl,-rpath,/tmp/catkin_workspace/devel_isolated/tf2_ros/lib:/tmp/catkin_workspace/devel_isolated/tf2/lib:/opt/ros/indigo/lib: 
```

Failed linking: 
```
Linking CXX executable /tmp/catkin_workspace/devel_isolated/tf2_ros/lib/tf2_ros/buffer_server
/usr/bin/cmake -E cmake_link_script CMakeFiles/tf2_ros_buffer_server.dir/link.txt --verbose=1
/usr/bin/c++       CMakeFiles/tf2_ros_buffer_server.dir/src/buffer_server_main.cpp.o  -o /tmp/catkin_workspace/devel_isolated/tf2_ros/lib/tf2_ros/buffer_server -rdynamic /tmp/catkin_workspace/devel_isolated/tf2_ros/lib/libtf2_ros.so /opt/ros/indigo/lib/libactionlib.so /opt/ros/indigo/lib/libmessage_filters.so /opt/ros/indigo/lib/libroscpp.so -lboost_signals -lboost_filesystem /opt/ros/indigo/lib/librosconsole.so /opt/ros/indigo/lib/librosconsole_log4cxx.so /opt/ros/indigo/lib/librosconsole_backend_interface.so -llog4cxx -lboost_regex /opt/ros/indigo/lib/libxmlrpcpp.so /tmp/catkin_workspace/install_isolated/lib/libtf2.so /opt/ros/indigo/lib/libroscpp_serialization.so /opt/ros/indigo/lib/librostime.so -lboost_date_time /opt/ros/indigo/lib/libcpp_common.so -lboost_system -lboost_thread -lpthread -lconsole_bridge -lboost_thread -lpthread -lconsole_bridge -Wl,-rpath,/tmp/catkin_workspace/devel_isolated/tf2_ros/lib:/tmp/catkin_workspace/install_isolated/lib:/opt/ros/indigo/lib: 
/tmp/catkin_workspace/devel_isolated/tf2_ros/lib/libtf2_ros.so: undefined reference to `tf2::BufferCore::allFramesAsYAML() const'
collect2: error: ld returned 1 exit status
```

tfoote, 2015-03-12:
I patched fix_84 for _allFramesAsDOT as well.

bricerebsamen, 2015-03-12:
yep that fixes it!

closed 2015-03-12 by tfoote

zhujimson, 2016-05-29:
hello,guys, I met this problem yesterday. When I want to run catkin_make command in catkin_ws path, it came out this error"undefined reference to `tf2::BufferCore::_allFramesAsDot() const' ", but it works fine before. And my system is indigo.

zhujimson, 2016-05-29:
After I saw  #84 . I still could not get any solution.

tfoote, 2016-05-29:
@zhujimson Please ask your question on answers.ros.org and provide the smallest example instructions to reproduce your linking problem in your question. This ticket has been closed for over a year.

PR #86 (2015-03-12): Use overloading instead of default arguments
This has better backwards compatibility for symbols. 
Fixes #84