ros/geometry2 #89: Successive sended transforms from a node not being received by tf2 python

marbosjo, 2015-03-18:
Hello,

we are porting the camera_pose_calibration (https://github.com/ros-perception/camera_pose) package to indigo and have found a bug in the tf2 python implementation. If two or more transforms are sended from the same node using successive calls to:
```python
tf2_ros.TransformBroadcaster().sendTransform()
```
Even with different TransformBroadcaster objects, only the last one is received by the tf core. If a delay is introduced by successive calls, all the transformations are received.

This bug does not happen in the tf2 cpp implementation neither in tf. I attach a minimal example covering the error, and the output shown by view_frames:

```python
#!/usr/bin/python
import tf2_ros
import rospy
from geometry_msgs.msg import TransformStamped

def send(pub, child_frame_id):
    transform = TransformStamped()
    transform.header.frame_id = 'world'
    transform.child_frame_id = child_frame_id
    transform.transform.translation.x = 0
    transform.transform.translation.y = 1
    transform.transform.translation.z = 2
    transform.transform.rotation.x = 0
    transform.transform.rotation.y = 0 
    transform.transform.rotation.z = 0 
    transform.transform.rotation.w = 1 
    transform.header.stamp = rospy.Time.now()
    pub.sendTransform(transform)

if __name__ == '__main__':
    rospy.init_node('tf2_timing_test')
    r = rospy.Rate(5)
    pub1 = tf2_ros.TransformBroadcaster()
    pub2 = tf2_ros.TransformBroadcaster()

    while not rospy.is_shutdown():
        send(pub1, 'frame1_from_pub1')
#        rospy.sleep(0.01);
        send(pub2, 'frame2_from_pub2')
#        rospy.sleep(0.01);
        send(pub1, 'frame3_from_pub1')
        r.sleep()
```

[image]

tfoote, 2015-03-19:
It looks like the c++ broadcaster as [a sending queue of 100](https://github.com/ros/geometry_experimental/blob/indigo-devel/tf2_ros/src/transform_broadcaster.cpp#L42)
while the python implementation has [a send queue of 1](https://github.com/ros/geometry_experimental/blob/indigo-devel/tf2_ros/src/tf2_ros/transform_broadcaster.py#L44)

The python should be made to match the c++ as this is causing a problem. It's a magic number and should probably be parameterized but just making them match would be better. 

I'll clean this up in a bit, but a pull-request would be appreciated.

mrath, 2015-03-20:
Fixed with queue_size 100 in pull request #90.

tfoote, 2015-03-20:
Fixed in #90

closed 2015-03-20 by tfoote