ros/geometry2 #92: crash

bricerebsamen, 2015-04-02:
I migrated some of my code to tf2 and it crashes from time to time. I am running the latest version from source. Here is the backtrace:

```
#0  0x00007ffff4852cc9 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff48560d8 in __GI_abort () at abort.c:89
#2  0x00007ffff488f394 in __libc_message (do_abort=do_abort@entry=1, 
    fmt=fmt@entry=0x7ffff499db28 "*** Error in `%s': %s: 0x%s ***\n")
    at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff489b66e in malloc_printerr (ptr=<optimized out>, 
    str=0x7ffff499dcf0 "double free or corruption (fasttop)", action=1)
    at malloc.c:4996
#4  _int_free (av=<optimized out>, p=<optimized out>, have_lock=0)
    at malloc.c:3840
#5  0x00007ffff6ed0146 in __gnu_cxx::new_allocator<std::pair<ros::Time, unsigned int> >::deallocate(std::pair<ros::Time, unsigned int>*, unsigned long) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#6  0x00007ffff6eccd40 in std::_Vector_base<std::pair<ros::Time, unsigned int>, std::allocator<std::pair<ros::Time, unsigned int> > >::_M_deallocate(std::pair<ros::Time, unsigned int>*, unsigned long) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#7  0x00007ffff6ecaa72 in std::vector<std::pair<ros::Time, unsigned int>, std::allocator<std::pair<ros::Time, unsigned int> > >::_M_insert_aux(__gnu_cxx::__normal_iterator<std::pair<ros::Time, unsigned int>*, std::vector<std::pair<ros::Time, unsigned int>, std::allocator<std::pair<ros::Time, unsigned int> > > >, std::pair<ros::Time, unsigned int> const&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#8  0x00007ffff6ec7d12 in std::vector<std::pair<ros::Time, unsigned int>, std::allocator<std::pair<ros::Time, unsigned int> > >::push_back(std::pair<ros::Time, unsigned int> const&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#9  0x00007ffff6ebd21b in tf2::BufferCore::getLatestCommonTime(unsigned int, unsigned int, ros::Time&, std::string*) const ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#10 0x00007ffff6ebf046 in tf2::BufferCore::testTransformableRequests() ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#11 0x00007ffff6ebb907 in tf2::BufferCore::setTransform(geometry_msgs::TransformStamped_<std::allocator<void> > const&, std::string const&, bool) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#12 0x00007ffff7201251 in tf2_ros::TransformListener::subscription_callback_impl(ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&, bool) () from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#13 0x00007ffff7201042 in tf2_ros::TransformListener::subscription_callback(ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#14 0x00007ffff72098ae in boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>::operator()(tf2_ros::TransformListener*, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) const ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#15 0x00007ffff7208d65 in void boost::_bi::list2<boost::_bi::value<tf2_ros::TransformListener*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>, boost::_bi::list1<boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&> >(boost::_bi::type<void>, boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>&, boost::_bi::list1<boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>&, int) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#16 0x00007ffff7208214 in void boost::_bi::bind_t<void, boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<tf2_ros::TransformListener*>, boost::arg<1> > >::operator()<boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> >(boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#17 0x00007ffff7207595 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<tf2_ros::TransformListener*>, boost::arg<1> > >, void, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#18 0x00007ffff7209afd in boost::function1<void, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>::operator()(boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) const ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#19 0x00007ffff7208f48 in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const>) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#20 0x00007ffff720b95a in boost::function1<void, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> >::operator()(boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const>) const ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#21 0x00007ffff720b156 in ros::SubscriptionCallbackHelperT<boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#22 0x00007ffff6194695 in ros::SubscriptionQueue::call() ()
   from /opt/ros/indigo/lib/libroscpp.so
#23 0x00007ffff61524f7 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/indigo/lib/libroscpp.so
#24 0x00007ffff6153303 in ros::CallbackQueue::callAvailable(ros::WallDuration)
    () from /opt/ros/indigo/lib/libroscpp.so
#25 0x00007ffff720289b in tf2_ros::TransformListener::dedicatedListenerThread()
    () from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#26 0x00007ffff720bec9 in boost::_mfi::mf0<void, tf2_ros::TransformListener>::operator()(tf2_ros::TransformListener*) const ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#27 0x00007ffff720bc46 in void boost::_bi::list1<boost::_bi::value<tf2_ros::TransformListener*> >::operator()<boost::_mfi::mf0<void, tf2_ros::TransformListener>, boost::_bi::list0>(boost::_bi::type<void>, boost::_mfi::mf0<void, tf2_ros::TransformListener>&, boost::_bi::list0&, int) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#28 0x00007ffff720b4c7 in boost::_bi::bind_t<void, boost::_mfi::mf0<void, tf2_ros::TransformListener>, boost::_bi::list1<boost::_bi::value<tf2_ros::TransformListener*> > >::operator()() ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#29 0x00007ffff720adcc in boost::detail::thread_data<boost::_bi::bind_t<void, boost::_mfi::mf0<void, tf2_ros::TransformListener>, boost::_bi::list1<boost::_bi::value<tf2_ros::TransformListener*> > > >::run() ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#30 0x00007ffff562aa4a in ?? ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#31 0x00007ffff5409182 in start_thread (arg=0x7fffd9c0a700)
    at pthread_create.c:312
#32 0x00007ffff491647d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

bricerebsamen, 2015-04-02:
Both this issue and #91 happen in the same program. The program creates a few callbacks (single threaded), a thread to do the heavy computation, and a transform listener. As a result the program contains 3 threads. I have a bunch of mutexes to protect my data, but I am assuming that the tf buffer is thread safe.

tfoote, 2015-04-02:
Can you provide any more info on what to do to reproduce this? It looks like it's crashing inside a push_back on a std::vector, which makes corruption much more likely.

bricerebsamen, 2015-04-02:
@tfoote it seems that we both posted at the same time. I am not sure you saw the description about what my program. And I just updated it. This is a very high level description and probably not enough information. I am not sure what else to say though.

When you say it makes corruption more likely what do you mean exactly? A stray pointer in my code that corrupts the tf buffer?

tfoote, 2015-04-02:
Yeah, I didn't see your comment. The tf buffer is supposed to be thread safe. I was mostly referring to it not being a double free. 

There are two possibilities, one is that the vector operations are not all properly protected with a mutex and two things are adding/removing iterating at the same time, or yeah it could be a complete smash where a pointer is walking into the wrong area. 

I'll dig in and find all the instances of operating on that vector and audit them for being properly protected.

bricerebsamen, 2015-04-02:
I got another crash:

```
Program received signal SIGSEGV, Segmentation Fault.
[Switching to Thread 0x7fffd9c09700 (LWP 26597)]
0x00007ffff562a66c in boost::detail::find_tss_data(void const*) ()
from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
```

Here is the backtrace for that thread:
```
#0  0x00007ffff562a66c in boost::detail::find_tss_data(void const*) ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#1  0x00007ffff562a6a9 in boost::detail::get_tss_data(void const*) ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#2  0x00007ffff61516ba in ros::CallbackQueue::setupTLS() ()
   from /opt/ros/indigo/lib/libroscpp.so
#3  0x00007ffff61530c0 in ros::CallbackQueue::callAvailable(ros::WallDuration)
    () from /opt/ros/indigo/lib/libroscpp.so
#4  0x00007ffff720289b in tf2_ros::TransformListener::dedicatedListenerThread()
    () from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#5  0x00007ffff720bec9 in boost::_mfi::mf0<void, tf2_ros::TransformListener>::operator()(tf2_ros::TransformListener*) const ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#6  0x00007ffff720bc46 in void boost::_bi::list1<boost::_bi::value<tf2_ros::TransformListener*> >::operator()<boost::_mfi::mf0<void, tf2_ros::TransformListener>, boost::_bi::list0>(boost::_bi::type<void>, boost::_mfi::mf0<void, tf2_ros::TransformListener>&, boost::_bi::list0&, int) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#7  0x00007ffff720b4c7 in boost::_bi::bind_t<void, boost::_mfi::mf0<void, tf2_ros::TransformListener>, boost::_bi::list1<boost::_bi::value<tf2_ros::TransformListener*> > >::operator()() ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#8  0x00007ffff720adcc in boost::detail::thread_data<boost::_bi::bind_t<void, boost::_mfi::mf0<void, tf2_ros::TransformListener>, boost::_bi::list1<boost::_bi::value<tf2_ros::TransformListener*> > > >::run() ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#9  0x00007ffff562aa4a in ?? ()
   from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.54.0
#10 0x00007ffff5409182 in start_thread (arg=0x7fffd9c09700)
    at pthread_create.c:312
#11 0x00007ffff491647d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

And for each thread:
```
Thread 15 (Thread 0x7fffbbfff700 (LWP 26669)):
#0  0x00007ffff6c45c98 in ?? () from /usr/lib/x86_64-linux-gnu/libgomp.so.1
#1  0x00007ffff6c44356 in ?? () from /usr/lib/x86_64-linux-gnu/libgomp.so.1
#2  0x00007ffff5409182 in start_thread (arg=0x7fffbbfff700)
    at pthread_create.c:312
#3  0x00007ffff491647d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

Thread 14 (Thread 0x7fffc0e89700 (LWP 26668)):
#0  0x00000000004260b0 in construct (this=0x7fffc0e88b00, 
    __val=@0x7fffc0e88adc: 12, __p=0x7fff980f9000)
    at /usr/include/c++/4.8/ext/new_allocator.h:130
#1  construct<float> (__a=..., __arg=@0x7fffc0e88adc: 12, __p=0x7fff980f9000)
    at /usr/include/c++/4.8/ext/alloc_traits.h:216
#2  push_back (__x=@0x7fffc0e88adc: 12, this=0x7fffc0e88b00)
    at /usr/include/c++/4.8/bits/stl_vector.h:905
#3  VelodyneLocalizer::alignStrength (this=0x7ffffffef7f0, dx=4, dy=8)
    at /home/brice/code/workspaces/merged/src/driving/packages/velodyne_localizer/src/velodyne_localizer.cpp:346
#4  0x000000000042649b in VelodyneLocalizer::computeBestOffset() [clone ._omp_fn.0] ()
    at /home/brice/code/workspaces/merged/src/driving/packages/velodyne_localizer/src/velodyne_localizer.cpp:397
#5  0x00007ffff6c4434a in ?? () from /usr/lib/x86_64-linux-gnu/libgomp.so.1
#6  0x00007ffff5409182 in start_thread (arg=0x7fffc0e89700)
    at pthread_create.c:312
#7  0x00007ffff491647d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

Thread 13 (Thread 0x7fffc168a700 (LWP 26667)):
#0  0x0000000000425fe7 in VelodyneLocalizer::alignStrength (
    this=0x7ffffffef7f0, dx=-4, dy=6)
    at /home/brice/code/workspaces/merged/src/driving/packages/velodyne_localizer/src/velodyne_localizer.cpp:340
#1  0x000000000042649b in VelodyneLocalizer::computeBestOffset() [clone ._omp_fn.0] ()
[... 175 lines cut ...]
#8  0x00007ffff6ebe876 in tf2::BufferCore::addTransformableRequest(unsigned int, std::string const&, std::string const&, ros::Time) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#9  0x0000000000441a69 in tf2_ros::MessageFilter<pcl::PointCloud<stdr_velodyne::PointType> >::add (this=0x7fffffff00e0, evt=...)
    at /home/brice/code/workspaces/merged/src/driving/external/geometry_experimental/tf2_ros/include/tf2_ros/message_filter.h:326
#10 0x0000000000437f0d in operator() (a0=..., this=0x6d1d58)
    at /usr/include/boost/function/function_template.hpp:767
#11 message_filters::CallbackHelper1T<ros::MessageEvent<pcl::PointCloud<stdr_velodyne::PointType> const> const&, pcl::PointCloud<stdr_velodyne::PointType> >::call (this=0x6d1d50, event=..., nonconst_force_copy=<optimized out>)
    at /opt/ros/indigo/include/message_filters/signal1.h:76
#12 0x0000000000437a3f in message_filters::Signal1<pcl::PointCloud<stdr_velodyne::PointType> >::call (this=<optimized out>, event=...)
    at /opt/ros/indigo/include/message_filters/signal1.h:119
#13 0x0000000000435ea6 in operator() (a0=..., this=0x6d2c98)
    at /usr/include/boost/function/function_template.hpp:767
#14 ros::SubscriptionCallbackHelperT<ros::MessageEvent<pcl::PointCloud<stdr_velodyne::PointType> const> const&, void>::call (this=0x6d2c90, params=...)
[... cut ...]
#17 0x00007ffff6153303 in ros::CallbackQueue::callAvailable(ros::WallDuration)
    () from /opt/ros/indigo/lib/libroscpp.so
#18 0x00007ffff61971b5 in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*)
    () from /opt/ros/indigo/lib/libroscpp.so
#19 0x00007ffff617f7bb in ros::spin() () from /opt/ros/indigo/lib/libroscpp.so
#20 0x0000000000423a6c in main (argc=1, argv=0x7fffffffd838)
    at /home/brice/code/workspaces/merged/src/driving/packages/velodyne_localizer/src/velodyne_localizer.cpp:750
```

bricerebsamen, 2015-04-02:
and for the original double free problem:

```
#0  0x00007ffff4852cc9 in __GI_raise (sig=sig@entry=6)
    at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ffff48560d8 in __GI_abort () at abort.c:89
#2  0x00007ffff488f394 in __libc_message (do_abort=do_abort@entry=1, 
    fmt=fmt@entry=0x7ffff499db28 "*** Error in `%s': %s: 0x%s ***\n")
    at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff489b66e in malloc_printerr (ptr=<optimized out>, 
    str=0x7ffff499dcf0 "double free or corruption (fasttop)", action=1)
    at malloc.c:4996
#4  _int_free (av=<optimized out>, p=<optimized out>, have_lock=0)
    at malloc.c:3840
#5  0x00007ffff6ed0146 in __gnu_cxx::new_allocator<std::pair<ros::Time, unsigned int> >::deallocate(std::pair<ros::Time, unsigned int>*, unsigned long) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#6  0x00007ffff6eccd40 in std::_Vector_base<std::pair<ros::Time, unsigned int>, std::allocator<std::pair<ros::Time, unsigned int> > >::_M_deallocate(std::pair<ros::Time, unsigned int>*, unsigned long) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#7  0x00007ffff6ecaa72 in std::vector<std::pair<ros::Time, unsigned int>, std::allocator<std::pair<ros::Time, unsigned int> > >::_M_insert_aux(__gnu_cxx::__normal_iterator<std::pair<ros::Time, unsigned int>*, std::vector<std::pair<ros::Time, unsigned int>, std::allocator<std::pair<ros::Time, unsigned int> > > >, std::pair<ros::Time, unsigned int> const&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#8  0x00007ffff6ec7d12 in std::vector<std::pair<ros::Time, unsigned int>, std::allocator<std::pair<ros::Time, unsigned int> > >::push_back(std::pair<ros::Time, unsigned int> const&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#9  0x00007ffff6ebd21b in tf2::BufferCore::getLatestCommonTime(unsigned int, unsigned int, ros::Time&, std::string*) const ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#10 0x00007ffff6ebf046 in tf2::BufferCore::testTransformableRequests() ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#11 0x00007ffff6ebb907 in tf2::BufferCore::setTransform(geometry_msgs::TransformStamped_<std::allocator<void> > const&, std::string const&, bool) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2.so
#12 0x00007ffff7201251 in tf2_ros::TransformListener::subscription_callback_impl(ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&, bool) () from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#13 0x00007ffff7201042 in tf2_ros::TransformListener::subscription_callback(ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#14 0x00007ffff72098ae in boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>::operator()(tf2_ros::TransformListener*, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) const ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#15 0x00007ffff7208d65 in void boost::_bi::list2<boost::_bi::value<tf2_ros::TransformListener*>, boost::arg<1> >::operator()<boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>, boost::_bi::list1<boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&> >(boost::_bi::type<void>, boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>&, boost::_bi::list1<boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>&, int) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#16 0x00007ffff7208214 in void boost::_bi::bind_t<void, boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<tf2_ros::TransformListener*>, boost::arg<1> > >::operator()<boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> >(boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#17 0x00007ffff7207595 in boost::detail::function::void_function_obj_invoker1<boost::_bi::bind_t<void, boost::_mfi::mf1<void, tf2_ros::TransformListener, ros::MessageEvent<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>, boost::_bi::list2<boost::_bi::value<tf2_ros::TransformListener*>, boost::arg<1> > >, void, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#18 0x00007ffff7209afd in boost::function1<void, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&>::operator()(boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&) const ()
   from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
#19 0x00007ffff7208f48 in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<tf2_msgs::TFMessage_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, [... cut ...]
#16 0x00007ffff6153303 in ros::CallbackQueue::callAvailable(ros::WallDuration)
    () from /opt/ros/indigo/lib/libroscpp.so
#17 0x00007ffff61971b5 in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*)
    () from /opt/ros/indigo/lib/libroscpp.so
#18 0x00007ffff617f7bb in ros::spin() () from /opt/ros/indigo/lib/libroscpp.so
#19 0x0000000000423a6c in main (argc=1, argv=0x7fffffffd838)
    at /home/brice/code/workspaces/merged/src/driving/packages/velodyne_localizer/src/velodyne_localizer.cpp:750
```

jhpanetta, 2015-04-07:
I found a possible "off the end of a vector" issue:  In tf2, BufferCore::getFrame reads:
```c++
tf2::TimeCacheInterfacePtr BufferCore::getFrame(CompactFrameID frame_id) const
{
  if (frame_id == 0 || frame_id > frames_.size()) /// @todo check larger values too
    return TimeCacheInterfacePtr();
  else
  {
    return frames_[frame_id];
  }
}
```
Since frames_ is an std::vector, this has the possibility of referencing beyond the end.  I'd say that the > should be >=.

bricerebsamen, 2015-04-07:
similarly here https://github.com/ros/geometry_experimental/blob/indigo-devel/tf2_ros/include/tf2_ros/message_filter.h#L367 shouldn't it be `message_count_ + 1 >= queue_size_`? Although this one will not lead to a segfault, it may result in an actual queue size on element larger than the requested queue size. It does not matter much for large queue sizes but would matter a great deal if you want the last message only (queue size of 1): in that case you'd get the one before latest message.

tfoote, 2015-04-16:
I've patched the getFrame indexing range check in 729a653 The message_filter.h queue size is correct as is. Adding one to the queue size has to overflow the queue to warrent poping values. As an example if the queue is empty and the queue is size 1. If it evaluates equal it will try to pop off from an empty queue which will not work. It could be `message_count_ >= queue_size_` without the `+1`

tfoote, 2015-04-17:
I've moved the vector from a member variable onto the stack. That member was not previously threadsafe. Please try thread fix#92 with commit fc80e5b

tfoote, 2015-04-17:
@bricerebsamen sorry forgot to ping you for the above

bricerebsamen, 2015-04-21:
I tested that patch without #101. It resulted in instant crash:
https://gist.github.com/bricerebsamen/7e1cb4bdbeaf0e376cb8

tfoote, 2015-04-21:
Shoot, I missed this comment when I merged #100.

tfoote, 2015-04-21:
Looking at that backtrace, we're in setTransform and the only lock in there is the frames_mutex_ (https://github.com/ros/geometry_experimental/blob/729a6531929203720f4a518ac9e79321fe3c5537/tf2/src/buffer_core.cpp#L248) which is a member of the class so should definitely not be null. 

```
#0  __GI___pthread_mutex_lock (mutex=0x0) at ../nptl/pthread_mutex_lock.c:66
#1  0x0000000000431ee8 in pthread_mutex_lock (m=<optimized out>) at /usr/include/boost/thread/pthread/mutex.hpp:61
#2  lock (this=<optimized out>) at /usr/include/boost/thread/pthread/mutex.hpp:113
#3  boost::unique_lock<boost::mutex>::lock (this=0x7fffd9cfb140) at /usr/include/boost/thread/lock_types.hpp:346
#4  0x00007ffff6ddae11 in boost::unique_lock<boost::mutex>::unique_lock(boost::mutex&) () from /home/brice/code/workspaces/merged/devel/lib/libtf2_ros.so
```

To me this looks like it's getting clobbered. I think that the reason for the change due to #100 is that it's changing the memory layout of the class.

wjwwood, 2015-04-22:
I had a look at it and I tend to agree with @tfoote that the tracebacks don't seem to be related to #100. Further more that #100 would not cause any memory corruption issues. The only thing that could tie #100 to the tracebacks here, imo, is if #100 some how exposed an existing bug in an unrelated part of the code. That being said, I think it is extremely unlikely that this is the case.

vrabaud, 2015-08-12:
@bricerebsamen , does that still happen with master ?

bricerebsamen, 2015-08-13:
I don't have the means to test that anymore, sorry

closed 2025-05-31 by sloretz