ros/geometry2 #102: tf.Exception thrown while using waitForTransform when a frame doesn't exist

marcoesposito1988, 2015-04-21:
As reported by other users as well in http://answers.ros.org/question/207039/tfexception-thrown-while-using-waitfortransform a bug seems to have appeared when using TF.

waitForTransform throws a tf.Exception if one of the frames doesn't exist yet (which is actually the point of waitForTransform).

It must be a bug in TF2. When I checked out geometry 1.11.4 the bug was still there, but when I also checked out geometry-experimental 0.5.7 the bug disappeared.

 Any idea about the reason?

sklaw, 2015-04-21:
Here are the combinations of TF and TF2 I've found so far that will cause this problem: 
TF 1.11.6 with TF2 0.5.9
TF 1.11.6 with TF2 0.5.8
TF 1.11.6 with TF2 0.5.7
TF 1.11.4 with TF2 0.5.9

tfoote, 2015-04-21:
What is the exception thrown? I added unit tests for wait for transform, in https://github.com/ros/geometry/commit/9624370b2327480ee429c5671455e3880825d256

tfoote, 2015-04-22:
Related questions on answers.ros.org
 * http://answers.ros.org/question/207582/does-tflookuptransform-python-take-time-to-fire-up/
 * http://answers.ros.org/question/207052/rospy-tf-waitfortransform-not-waiting/

tfoote, 2015-04-22:
I believe I've found the issue in this commit: https://github.com/ros/geometry_experimental/commit/2f5ba1db9c86043221d49d25683cb2ef82211f1e

When running inside rospy, the roscpp client is not initialized and thus ros::ok is false. I'm going to have to think about how to make this work cleanly in both roscpp and rospy. .

tfoote, 2015-04-22:
This should be fixed in geometry_experimental 0.5.11

commit 7284788f (2015-04-22): do not short circuit waitForTransform timeout when running inside pytf. Fixes #102

roscpp is not initialized inside pytf which means that ros::ok is not
valid. This was causing the timer to abort immediately.

This breaks support for pytf with respect to early breaking out of a loop re #26.
This is conceptually broken in pytf, and is fixed in tf2_ros python implementation.
If you want this behavior I recommend switching to the tf2 python bindings.
closed 2015-04-22 by tfoote