ros/geometry2 #103: tf2_ros: transform in python doesn't work

jobstvogt, 2015-04-22:
I think the transform-Method of buffer_interface.py doesn't work or can someone show me how to use it?

I need something like transformPose from tf.listener to transform a PoseStamped but with tf2...

As a first solution i have written a helper-method that uses the transform from tfBuffer.lookup_transform to transform the PoseStamped but shouldn't there be a method in tf2_ros for that?

tfoote, 2015-04-22:
The transform methods for PoseStamped are defined in [tf2_geometry_msgs](http://wiki.ros.org/tf2_geometry_msgs) In tf2 we use a dependency injection method to not require datatype access in the core.   There's an example using PointStamped you can follow [here](https://github.com/ros/geometry_experimental/blob/indigo-devel/test_tf2/test/test_buffer_client.py)

In general please ask questions on answers.ros.org not as a ticket.

closed 2015-04-22 by tfoote

jobstvogt, 2015-04-23:
I'm sorry, thank you very much!