ros/geometry2 #108: rviz tf display tool does not handle static transforms correctly

davetcoleman, 2015-05-09:
Seems to me static transforms should always display in Rviz, but instead they fade out as if they were non-static. Resetting the tf display makes the transform show again until the "Frame Timeout" is reached. Thoughts?

tfoote, 2015-05-10:
The rviz display needs to be updated to know the difference between regular and static transforms. For efficiency static_transforms are now published as latched topics. The gray out logic in rviz's display does not take that into account. This should be filed against the rviz tf plugin.

closed 2015-05-10 by tfoote

davetcoleman, 2015-05-10:
Sure, i just thought the issue would be seen by the right person (you) on this issue tracker

https://github.com/ros-visualization/rviz/issues/888

tfoote, 2015-05-10:
Yeah, thanks for refiling.