ros/geometry2 #110: add getYaw helper function

tfoote, 2015-05-17:
It saves several lines of code which could easily be done wrong.

vrabaud, 2015-05-17:
I can make a PR but where do you see that ? a member function of the tf2::Quaternion object ? A standalone function ? (that could work with tf2::Matrix3x3 and tf2::QUaternion)

tfoote, 2015-05-17:
I like the bare function. It would be nice to support extendable datatypes kind of like convert does with the inclusion of the appropriate headers. It doesn't need to be templated as the datatypes can just be overloaded. As a bare function we can also provide it for the message datatypes. 

It should probably be something like getHeading for a clearer scope of what it's providing. 

I guess it should get it's own header file for clarity. Are there other similar methods missing? Common, simple, but painful?

vrabaud, 2015-05-17:
well, there are a few other functions but they are member functions (like setIdentity for a transform).
Should that function be put in https://github.com/ros/geometry_experimental/blob/indigo-devel/tf2/include/tf2/convert.h ? It's already a bit of everything in there and those functions are supposed to work for any datatypes.

tfoote, 2015-05-17:
convert.h does seem like a reasonable place. And it can be added side by side with the toMsg fromMsg in the other packages too.

vrabaud, 2015-05-17:
#112 might be a solution

mikeferguson, 2015-05-17:
+1 to having a getYaw -- especially one that accepts the orientation of geometry_msgs/Pose directly

vrabaud, 2015-05-17:
hmm, with #112, it only gives the yaw of anything that can be converted to a tf2::Quaternion. Hence, not a pose. You can just call getYaw(pose.orientation). We could also specialize it for a Pose. I just feel that we should keep it simple. If you feel it should work for several types, please let me know which. I see Transform and Pose. Anything else ?

Just saw that my PR would not work for Stamped messages, will fix that.

mikeferguson, 2015-05-17:
pose.orientation is fine -- I really just want to avoid converting from message to TF to yaw (something I find is an often repeated calculation)

tfoote, 2015-06-08:
Fixed in #112

closed 2015-06-08 by tfoote