ros/geometry2 #116: Missing convertions

mistoll, 2015-07-30:
Once understood the convert method is quite helpfull. But IMO there are some basic convertion methods missing. 

For example:
tf2_geometry_msgs.h missing Vector3, Pose, Point (only stamped versions available)
tf2_eigen.h missing Quaterniond, Matrix3d

Furthermore the documentation on creating convertion methods is incomplete or hard to find.

vrabaud, 2015-08-13:
right, this documentation needs advertisement, like on the wiki.
You mention a few types, but what would they convert to ? Thx

mistoll, 2015-08-17:
Nice convertions would be:
Eigen::Quaterniond <=> geometry_msgs::Quaternion
Eigen::Vector3 <=> geometry_msgs::Vector, geometry_msgs::Point
Eigen::Matrix3d <=> geometry_msgs::Matrix
Also great (but might be incompatible)
Eigen::Matrixxd <=> geometry_msgs covariances (e.g. float64[36])

vrabaud, 2015-08-17:
Ok, if you have any other types that you'd like to have converted, please add in here.

Eigen::Vector3 <=> geometry_msgs::Vector, geometry_msgs::Point

is a tricky one: a vector is not the same things as a point in ROS (a vector is a direction, that is therefore not translated when a transform is applied to it, cf https://github.com/ros/geometry_experimental/blob/jade-devel/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h#L77 )

Same with the matrix and the covariance matrix: the info is the same, but not the semantic (same with pose and transform).

Anyway, I need to write the wiki page about conversion and utils.h, and add those conversions above that make sense.

mistoll, 2015-08-17:
I wasn't aware of the doTransform method. That complicates things.

So what do you recommend for those non-bijective mappings? Should everyone implement them application-specific or not use convert at all?

tfoote, 2015-08-17:
If it's not bijective it shouldn't use the convert method. You should
define a method which captures the semantic meaning, potentially
application specific, potentially somewhat generic and shareable between
similar applications as any function can be done already.

cmansley, 2015-12-10:
I agree with @tfoote, but we should define which conversions are the bijective ones and which are not. Because `geometry_msgs::Pose` is being converted into `tf2::Transform` in tf2_geometry_msgs.h, but it could have easily been `geometry_msgs::Transform` instead. 

And these conversion functions are replicated in the buffer_core.cpp.

So, I guess there are two things. 
* When do we use the convert interface defined in convert.h?
* What is the policy for defining "custom" methods?

tfoote, 2017-07-14:
#202 adds the primatives for geometry_msgs

closed 2025-05-31 by sloretz