ros/geometry2 #118: tf2_sensor_msgs depends on Eigen headers, but does not add them to include path

mikeferguson, 2015-08-09:
On Indigo/14.04: using tf2_sensor_msgs in a package that does not directly depend on Eigen, I get the following error:

```
/opt/ros/indigo/include/tf2_sensor_msgs/tf2_sensor_msgs.h:36:23: fatal error: Eigen/Eigen: No such file or directory
 #include <Eigen/Eigen>
```

If I then find_package(Eigen) and add the EIGEN_INCLUDE_DIRS, all compiles well. Should the EIGEN_INCLUDE_DIRS be added to the catkin_package(INCLUDE_DIRS ...)? or is there another way to specify this depend without directly having each package need to add Eigen to the CMake.

vrabaud, 2015-08-12:
```catkin_package(INCLUDE_DIRS ...)``` is only for build time.
Everything seems there on the CMake side: https://github.com/ros/geometry_experimental/blob/jade-devel/tf2_sensor_msgs/CMakeLists.txt#L11

You are using a deb package and when looking at /opt/ros/indigo/share/tf2_sensor_msgs/cmake/tf2_sensor_msgsConfig.cmake, do you have ```/usr/include/eigen3``` in there ?

mikeferguson, 2015-08-12:
Ok, it does appear that the tf2_sensor_msgsConfig.cmake has eigen3 in it... not really sure what is going on then.. and now I can no longer reproduce the error, so there must have been something wonky in my environment at the time. Sorry for the noise.

closed 2015-08-12 by mikeferguson