ros/geometry2 #124: transforming a Vector3 in python applies the translation

bricerebsamen, 2015-09-23:
according to the doc:
```
# This represents a vector in free space. 
# It is only meant to represent a direction. Therefore, it does not
# make sense to apply a translation to it (e.g., when applying a 
# generic rigid transformation to a Vector3, tf2 will only apply the
# rotation). If you want your data to be translatable too, use the
# geometry_msgs/Point message instead.
```
(from http://docs.ros.org/jade/api/geometry_msgs/html/msg/Vector3.html)

the following code shows that it's not the case:
```
t = TransformStamped()
t.transform.translation.z = 1
t.transform.rotation.w = 1

v = Vector3Stamped()
v.vector.x = 1
v.vector.y = 0
v.vector.z = 0

vt = tf2_geometry_msgs.do_transform_vector3(v, t)
```
As a result, `vt.z` is 1, so the translation is applied.

bricerebsamen, 2015-09-23:
This program can be used to test:
```
#!/usr/bin/python

import math
import numpy as np
import roslib; roslib.load_manifest('tf2_geometry_msgs')
import rospy
import tf
import tf2_ros
import tf2_geometry_msgs
from geometry_msgs.msg import TransformStamped, Vector3Stamped

def test_similar(test_name, v, x, y, z):
    tolerance = 1e-6
    if np.abs(np.array([v.x, v.y, v.z]) - np.array([x, y, z])).max() > tolerance:
        print 'FAIL:', test_name
        print 'expected:', x, y, z
        print 'actual  :', v.x, v.y, v.z

# a translation should not modify a Vector3 (only rotation should)
# testing with a transform that is a pure translation
t = TransformStamped()
t.transform.translation.z = 1
t.transform.rotation.w = 1

v = Vector3Stamped()
v.vector.x = 1
v.vector.y = 0
v.vector.z = 0

vt = tf2_geometry_msgs.do_transform_vector3(v, t)
test_similar('translation should not be applied', vt.vector, v.vector.x, v.vector.y, v.vector.z)

# making sure the rotation is applied properly
t = TransformStamped()
t.transform.translation.x = 0
q = tf.transformations.quaternion_from_euler(0, 0, math.radians(90))
t.transform.rotation.x = q[0]
t.transform.rotation.y = q[1]
t.transform.rotation.z = q[2]
t.transform.rotation.w = q[3]

v = Vector3Stamped()
v.vector.x = 1
v.vector.y = 0
v.vector.z = 0

vt = tf2_geometry_msgs.do_transform_vector3(v, t)
test_similar('rotation', vt.vector, 0, 1, 0)
```

bricerebsamen, 2015-09-23:
Note: I tested under `indigo`, I haven't tested with `jade`

tfoote, 2015-12-07:
Thanks for the report. This definitely looks like a bug.

closed 2015-12-11 by tfoote

mdelp-es, 2021-11-03:
cause its considering it to be a point?

PR #128 (2015-12-09): Vector3 not affected by translation in a transform
Fixes #124 
Connects to #130