ros/geometry2 #138: tf2_ros static_transform publisher does not check for non unitary quaternions

g-gemignani, 2015-12-22:
I believe the node should check if the angles passed in input are Euler angles or quaternions. If quaternions it should check if they are unitary or not.

tfoote, 2015-12-22:
Can you provide a case where it does not differentiate? It checks the number of arguments to differentiate euler angles vs quaternions.

Checking the normalization of the quaternion would be a good idea a PR to add a validation in util.h and use it in the static transformer program would be appreciated.

g-gemignani, 2015-12-23:
As requested I created a PR

closed 2015-12-23 by g-gemignani

tfoote, 2015-12-24:
@g-gemignani can you link to the PR? I do not see it at the moment.

g-gemignani, 2015-12-24:
@tfoote here is the link. I committed on my previous PR. Hope it's not a problem. 

https://github.com/ros/geometry_experimental/pull/133