ros/geometry2 #145: Compiler error due to not found message headers

firesurfer, 2016-01-15:
Hey,
I tried using the TransformBroadcaster in my own program (c++). 
After adding the include
```c++
#include <tf2_ros/transform_broadcaster.h>
```

I got the following errors from the transform_broadcast.h file:
First it couldnt find the #include "geometry_msgs/msg/transform_stamped.h"
I solved this by changing the include to:
```c++
#include "geometry_msgs/msg/transform_stamped.hpp"
```
Second it couldnt find the tf2_msgs::msg::TFMessage
I fixed this by adding the include
```c++
#include "tf2_msgs/msg/tf_message.hpp"
```

tfoote, 2016-01-15:
Can you clarify where and how and what version you're using?

firesurfer, 2016-01-15:
I'm using the version that comes with ros2 alpha 3. 

How: I'm having a project where I include the file:
```c++
#include <tf2_ros/transform_broadcaster.h>
```

Error:
I get this error and some more errors all complaining about that it can't find tf2_msgs
```
~/workspace/ros_ws/install/include/tf2_ros/transform_broadcaster.h:72:32: error: ‘tf2_msgs’ was not declared in this scope
   rclcpp::publisher::Publisher<tf2_msgs::msg::TFMessage>::SharedPtr publisher_;
```

Solution:

In the file: ~/workspace/ros_ws/install/include/tf2_ros/transform_broadcaster.h add the following line:
```c++
#include "tf2_msgs/msg/tf_message.hpp"
```

What is kind of strange that I can't reproduce the error that says it can't find the geometry messages. I think I did something wrong myself there....

tfoote, 2016-01-15:
Thanks, yes it should be .hpp

The .h files are generated for C, you probably now have C support in your workspace so the include finds a file but doesn't have the right definitions.

tfoote, 2016-02-17:
fixed in #146

PR #146 (2016-01-21): Fix #include of transform_stamped message
It should be a .hpp not .h now. 

Fixes #145
commit 98c5e7d4 (2016-01-21): Fix #include of transform_stamped message

It should be a .hpp not .h now. 

Fixes #145
closed 2016-02-17 by tfoote