ros/geometry2 #156: [ros2] tf2_monitor does not output any broadcasters or frames

wjwwood, 2016-02-17:
To reproduce (using alpha 4 rc extracted to `~/ros2`):

- `source ~/ros2/setup.bash`
- `~/ros2/bin/tf2_ros/static_transform_publisher 0 0 1 0 0 0 1 foo bar`
- `./tf2_monitor foo bar` or just `./tf2_monitor`

This is what I get:

```
LOOPING due to no latching at the moment
Waiting for transform chain to become available between foo and bar
LOOPING due to no latching at the moment
LOOPING due to no latching at the moment

RESULTS: for foo to bar
Chain is: foo -> bar
Net delay     avg = 2.76988e+08: max = 1.45574e+09

Frames:

All Broadcasters:
LOOPING due to no latching at the moment
LOOPING due to no latching at the moment

RESULTS: for foo to bar
Chain is: foo -> bar
Net delay     avg = 5.01272e+08: max = 1.45574e+09

Frames:

All Broadcasters:
LOOPING due to no latching at the moment
LOOPING due to no latching at the moment
```

A similar set of actions in ROS 1 gets me something like this:

```
RESULTS: for all Frames

Frames:
Frame: foo published by unknown_publisher Average Delay: 0.000242583 Max Delay: 0.000801

All Broadcasters:
Node: unknown_publisher 15841.4 Hz, Average Delay: 0.000242583 Max Delay: 0.000801

RESULTS: for all Frames

Frames:
Frame: foo published by unknown_publisher Average Delay: 0.000469953 Max Delay: 0.002289

All Broadcasters:
Node: unknown_publisher 19420.9 Hz, Average Delay: 0.000469953 Max Delay: 0.002289
```

Also the reported time from the `tf2_monitor` gives large net delays, which might also be a bug or it might be related to us having to loop rather than latch on the static transform (for now).

gerkey, 2016-04-05:
By contrast, `tf2_echo` seems to work fine.

Karsten1987, 2016-10-04:
tf2_echo crashes without any message on windows

jonbinney, 2016-11-01:
Is tf2_monitor only a ros2 thing? I don't see it in tf2_ros in kinetic.

tfoote, 2016-11-01:
The tool was not ported during the migration since it's a command line tool and is using tf2 under the hood. Not changing the entry point meant we didn't need to update documentation. We needed it when prototyping for ros2 since we were only porting geometry2. It would probably make sense to migrate it now. If it migrates it should go into tf2_tools.

tfoote, 2017-07-14:
I've opened a ticket for tf2_monitor in the ros2 repo: https://github.com/ros2/geometry2/issues/37 closing here.

closed 2017-07-14 by tfoote