ros/geometry2 #160: promote geometry_experimental to geometry2

tfoote, 2016-03-21:
We should take this opportunity to review the documentation and make sure it's at parity with geometry. Below are the tasks from my audit. 

Validate tutorials and documentation have coverage of everything in geometry.
* [ ] Port tf/Overview to tf2/Overview
* [ ] Port tf/FAQ
* [ ] Port geometry/RotationMethods
* [ ] Port geometry/CoordinateFrameConventions (reference REPs 103, 105, 120, remove tf_prefix)

* [ ] Rename geometry_experimental -> geometry2

* [ ] Create wiki page to describe templated convert and transform methods:
 * [ ] Top level documentation not just quick tutorial with links: http://wiki.ros.org/tf2/Tutorials/Migration/DataConversions
 * [ ] Document how to add a new package implementation

Fill in wiki pages for:
* [ ] tf2_bullet (mostly link/include of convert documentation)
* [ ] tf2_eigen (mostly link/include of convert documentation)
* [ ] tf2_geometry_msgs (mostly link/include of convert documentation)
* [ ] tf2_kdl (mostly link/include of convert documentation)
* [ ] tf2_msgs (document as internal APIs) 
* [ ] tf2_py (document as python bridge, internal API) 
* [ ] tf2_ros 
* [ ] tf2_sensor_msgs (mostly link/include of convert documentation)
* [ ] tf2_tools
* [ ] geometry2

Much of the content can be shared using includes

tf2_ros
* [ ] Fill in doxygen main page for c++ giving API overview
* [ ] Fill in sphinx home page to give API overview

Renaming
* [ ] Find all references and update them to geometry2

jacquelinekay, 2016-03-21:
I can help out with this.

tfoote, 2016-03-31:
Everything is checked off closing. Thanks for all the help @jacquelinekay @mikaelarguedas

closed 2016-03-31 by tfoote