ros/geometry2 #161: Add `gazebo::math::` to `geometry_msgs::` conversions

alextoind, 2016-03-21:
In my project I am dealing with tf2 conversion from `gazebo::math::` to `geometry_msgs::` types. I have already implemented some methods Iike the following: 

    namespace tf2 {
    
    inline
    geometry_msgs::TransformStamped toMsg(const tf2::Stamped<gazebo::math::Pose> &in)
    {
      geometry_msgs::TransformStamped msg;
      msg.header.frame_id = in.frame_id_;
      msg.header.stamp = in.stamp_;
      msg.transform.translation.x = in.pos.x;
      msg.transform.translation.y = in.pos.y;
      msg.transform.translation.z = in.pos.z;
      msg.transform.rotation.w = in.rot.w;
      msg.transform.rotation.x = in.rot.x;
      msg.transform.rotation.y = in.rot.y;
      msg.transform.rotation.z = in.rot.z;
      return msg;
    }
    
    } // namespace tf2

If it is a good idea to add these features to the library, I would like to know where should be the correct place to add this code for a pull request. Is there some reason not to do so?

tfoote, 2016-03-21:
The tf2 conversion methods are designed to be provided by separate packages, for example `tf2_bullet` and `tf2_kdl`provide the conversions for bullet and kdl.  These functions would work well in a `tf2_gazebo` package probably in the gazebo_ros_pkgs repository. https://github.com/ros-simulation/gazebo_ros_pkgs

tfoote, 2017-07-10:
Since the conversions make more sense in gazebo_ros_pkgs I'm going to close the ticket on this repo.

@j-rivero fyi this could be a good enhancement for gazebo_ros_pkgs

closed 2017-07-10 by tfoote

j-rivero, 2017-07-10:
@alextoind feel free to copy the issue to the gazebo_ros_pkgs repo if you are still interested.

alextoind, 2017-07-12:
@j-rivero actually I'm not intereseted in this feature anymore. I think that it is not really useful if nobody but me asked for such behavior. :roll_eyes: