ros/geometry2 #166: Error processing tf2 packages

onjen, 2016-03-31:
Tried to install Jade today, after executing
```shell
wstool init -j8 src jade-desktop-full-wet.rosinstall
```

I had the following errors:

```shell
Exception caught during install: Error processing 'geometry_experimental/tf2' : [geometry_experimental/tf2] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2 failed.
Error processing 'geometry_experimental/tf2_geometry_msgs' : [geometry_experimental/tf2_geometry_msgs] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2_geometry_msgs/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2_geometry_msgs-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2_geometry_msgs failed.
Error processing 'geometry_experimental/tf2_kdl' : [geometry_experimental/tf2_kdl] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2_kdl/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2_kdl-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2_kdl failed.
Error processing 'geometry_experimental/tf2_msgs' : [geometry_experimental/tf2_msgs] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2_msgs/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2_msgs-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2_msgs failed.
Error processing 'geometry_experimental/tf2_py' : [geometry_experimental/tf2_py] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2_py/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2_py-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2_py failed.
Error processing 'geometry_experimental/tf2_ros' : [geometry_experimental/tf2_ros] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2_ros/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2_ros-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2_ros failed.

ERROR in config: Error processing 'geometry_experimental/tf2' : [geometry_experimental/tf2] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2 failed.
Error processing 'geometry_experimental/tf2_geometry_msgs' : [geometry_experimental/tf2_geometry_msgs] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2_geometry_msgs/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2_geometry_msgs-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2_geometry_msgs failed.
Error processing 'geometry_experimental/tf2_kdl' : [geometry_experimental/tf2_kdl] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2_kdl/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2_kdl-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2_kdl failed.
Error processing 'geometry_experimental/tf2_msgs' : [geometry_experimental/tf2_msgs] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2_msgs/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2_msgs-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2_msgs failed.
Error processing 'geometry_experimental/tf2_py' : [geometry_experimental/tf2_py] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2_py/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2_py-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2_py failed.
Error processing 'geometry_experimental/tf2_ros' : [geometry_experimental/tf2_ros] Checkout of https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2_ros/0.5.13-0.tar.gz version geometry_experimental-release-release-jade-tf2_ros-0.5.13-0 into /home/johannes/ros_catkin_ws/src/geometry_experimental/tf2_ros failed.
```

Yesterday the download worked fine. Same issue in Indigo.

tfoote, 2016-03-31:
Apparently curl does not follow the redirects needed until this is merged: https://github.com/ros/rosdistro/pull/10926

```
$ curl -v https://github.com/ros-gbp/geometry_experimental-release/archive/release/jade/tf2/0.5.13-0.tar.gz -o /tmp/0.5.13-0.tar.gz
* Hostname was NOT found in DNS cache
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0*   Trying 192.30.252.131...
* Connected to github.com (192.30.252.131) port 443 (#0)
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs
* SSLv3, TLS handshake, Client hello (1):
} [data not shown]
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0* SSLv3, TLS handshake, Server hello (2):
{ [data not shown]
* SSLv3, TLS handshake, CERT (11):
{ [data not shown]
* SSLv3, TLS handshake, Server key exchange (12):
{ [data not shown]
* SSLv3, TLS handshake, Server finished (14):
{ [data not shown]
* SSLv3, TLS handshake, Client key exchange (16):
} [data not shown]
* SSLv3, TLS change cipher, Client hello (1):
} [data not shown]
* SSLv3, TLS handshake, Finished (20):
} [data not shown]
* SSLv3, TLS change cipher, Client hello (1):
{ [data not shown]
* SSLv3, TLS handshake, Finished (20):
{ [data not shown]
* SSL connection using ECDHE-RSA-AES128-GCM-SHA256
* Server certificate:
* 	 subject: businessCategory=Private Organization; 1.3.6.1.4.1.311.60.2.1.3=US; 1.3.6.1.4.1.311.60.2.1.2=Delaware; serialNumber=5157550; street=88 Colin P Kelly, Jr Street; postalCode=94107; C=US; ST=California; L=San Francisco; O=GitHub, Inc.; CN=github.com
* 	 start date: 2016-03-10 00:00:00 GMT
* 	 expire date: 2018-05-17 12:00:00 GMT
* 	 subjectAltName: github.com matched
* 	 issuer: C=US; O=DigiCert Inc; OU=www.digicert.com; CN=DigiCert SHA2 Extended Validation Server CA
* 	 SSL certificate verify ok.
> GET /ros-gbp/geometry_experimental-release/archive/release/jade/tf2/0.5.13-0.tar.gz HTTP/1.1
> User-Agent: curl/7.35.0
> Host: github.com
> Accept: */*
> 
< HTTP/1.1 301 Moved Permanently
* Server GitHub.com is not blacklisted
< Server: GitHub.com
< Date: Thu, 31 Mar 2016 18:08:13 GMT
< Content-Type: text/html; charset=utf-8
< Transfer-Encoding: chunked
< Status: 301 Moved Permanently
< Cache-Control: no-cache
< Vary: X-PJAX
< Location: https://github.com/ros-gbp/geometry2-release/archive/release/jade/tf2/0.5.13-0.tar.gz
< X-UA-Compatible: IE=Edge,chrome=1
< X-Request-Id: bcc48a6d286d3e79dcb837325a1348b1
< X-Runtime: 0.013513
< Content-Security-Policy: default-src 'none'; base-uri 'self'; block-all-mixed-content; child-src render.githubusercontent.com; connect-src 'self' uploads.github.com status.github.com api.github.com www.google-analytics.com github-cloud.s3.amazonaws.com api.braintreegateway.com client-analytics.braintreegateway.com wss://live.github.com; font-src assets-cdn.github.com; form-action 'self' github.com gist.github.com; frame-ancestors 'none'; frame-src render.githubusercontent.com; img-src 'self' data: assets-cdn.github.com identicons.github.com www.google-analytics.com collector.githubapp.com *.gravatar.com *.wp.com checkout.paypal.com *.githubusercontent.com; media-src 'none'; object-src assets-cdn.github.com; plugin-types application/x-shockwave-flash; script-src assets-cdn.github.com; style-src 'unsafe-inline' assets-cdn.github.com
< Strict-Transport-Security: max-age=31536000; includeSubdomains; preload
< Public-Key-Pins: max-age=1209600; pin-sha256="WoiWRyIOVNa9ihaBciRSC7XHjliYS9VwUGOIud4PB18="; pin-sha256="RRM1dGqnDFsCJXBTHky16vi1obOlCgFFn/yOhI/y+ho="; pin-sha256="k2v657xBsOVe1PQRwOsHsw3bsGT2VzIqz5K+59sNQws="; pin-sha256="K87oWBWM9UZfyddvDfoxL+8lpNyoUB2ptGtn0fv6G2Q="; pin-sha256="IQBnNBEiFuhj+8x6X8XLgh01V9Ic5/V3IRQLNFFc7v4="; pin-sha256="iie1VXtL7HzAMF+/PVPR9xzT80kQxdZeJ+zduCB3uj0="; pin-sha256="LvRiGEjRqfzurezaWuj8Wie2gyHMrW5Q06LspMnox7A="; includeSubDomains
< X-Content-Type-Options: nosniff
< X-Frame-Options: deny
< X-XSS-Protection: 1; mode=block
< Vary: Accept-Encoding
< X-Served-By: 3f38dada85f97412f7f824e59f77fa9d
< X-GitHub-Request-Id: 4623323A:5F9B:230BC26C:56FD678D
< 
{ [data not shown]
100   151    0   151    0     0    299      0 --:--:-- --:--:-- --:--:--   299
* Connection #0 to host github.com left intact
```

tfoote, 2016-03-31:
I've opened a ticket on vcstools to accept the redirect: https://github.com/vcstools/vcstools/issues/114 in the mean time it should be merged now and not a problem using the new urls.

tfoote, 2016-03-31:
Please comment back if you're still having trouble and we can reopen.

closed 2016-03-31 by tfoote

onjen, 2016-04-01:
Download worked fine now, thanks!

tfoote, 2016-08-25:
If others run into this in the future. There's a workaround sed script example at: https://github.com/vcstools/vcstools/issues/114 but the tarball based rosinstalls will not work when a repo gets renamed even with the redirects.