ros/geometry2 #172: tf2_msgs: NO_ERROR creates a name clash on Windows with MSVC

ibadr, 2016-05-04:
The <tt>TF2Error.msg</tt> file defines on its first line ``uint8 NO_ERROR = 0``. The generated C++ code for this message in <tt>include/tf2_msgs/TF2Error.h</tt> includes the following line
```
enum { NO_ERROR = 0u };
```
This line causes a name clash on Windows with MSVC and MinGW, since ``NO_ERROR`` is already defined to be a constant 0. A simple fix would be to conditionally undefine ``NO_ERROR`` on Windows in the generated C++ code, e.g. as done [here](https://github.com/vinipsmaker/tufao/issues/4) and [here](http://www.netmite.com/android/mydroid/frameworks/base/include/utils/Errors.h).

The relevant part of the compilation error on Windows using MSVC and NMake is below
```
C:\Users\islam\ws_winros\ws\devel\include\tf2_msgs/TF2Error.h(78) : error C2059: syntax error : 'constant'
        C:\Users\islam\ws_winros\ws\devel\include\tf2_msgs/TF2Error.h(90) : see reference to class template instantiation 'tf2_msgs::TF2Error_<ContainerAllocator>' being compiled
C:\Users\islam\ws_winros\ws\devel\include\tf2_msgs/TF2Error.h(78) : error C2143: syntax error : missing ';' before '}'
C:\Users\islam\ws_winros\ws\devel\include\tf2_msgs/TF2Error.h(78) : error C2238: unexpected token(s) preceding ';'
C:\Users\islam\ws_winros\ws\src\tf2\src\buffer_core.cpp(597) : error C2589: 'constant' : illegal token on right side of '::'
C:\Users\islam\ws_winros\ws\src\tf2\src\buffer_core.cpp(597) : error C2059: syntax error : '::'
C:\Users\islam\ws_winros\ws\src\tf2\src\buffer_core.cpp(598) : error C2143: syntax error : missing ';' before '{'
C:\Users\islam\ws_winros\ws\src\tf2\src\buffer_core.cpp(601) : error C2039: 'CONNECTIVITY_ERROR' : is not a member of 'tf2_msgs::TF2Error_<std::allocator<void>>'
C:\Users\islam\ws_winros\ws\src\tf2\src\buffer_core.cpp(601) : error C2065: 'CONNECTIVITY_ERROR' : undeclared identifier
C:\Users\islam\ws_winros\ws\src\tf2\src\buffer_core.cpp(601) : error C2051: case expression not constant
C:\Users\islam\ws_winros\ws\src\tf2\src\buffer_core.cpp(603) : error C2039: 'EXTRAPOLATION_ERROR' : is not a member of 'tf2_msgs::TF2Error_<std::allocator<void>>'
C:\Users\islam\ws_winros\ws\src\tf2\src\buffer_core.cpp(603) : error C2065: 'EXTRAPOLATION_ERROR' : undeclared identifier
C:\Users\islam\ws_winros\ws\src\tf2\src\buffer_core.cpp(603) : error C2051: case expression not constant
```

tfoote, 2017-07-10:
We're compiling for Windows in the ros2 prototypes and I don't believe that we've run into this issue. Is it enough to do this just in the header? Or are we just luck that it's defined the same in the implementation?

tfoote, 2017-07-12:
Looking into this, it's the generated code from the msg. And our usage is properly scoped. There's already a large code base using this so we can't easily switch.

If this is something that continues to be a problem we will have to patch the code generator to do the undef. But for now since this is known to work on Windows in most cases it might make sense to undef this anytime you use a header that brings in WinError.h or make sure that the message file is included before WinError.h.

I'm going to close this as I think it's out of scope for fixing in this repository.

closed 2017-07-12 by tfoote

serge-nikulin, 2017-10-26:
This issue happens on my PC when a Windows SDK is installed.  

The build issue disappeared when the SDK has been uninstalled.  

It's OK to leave this unfixed but then ROS2 can't claim formal Windows compatibility since Windows SDK is essential for Windows GUI development.