ros/geometry2 #174: TF2 ROS independence

phil0stine, 2016-05-10:
In the [tf2 design doc](http://wiki.ros.org/tf2/Design), tf2 core is said to be ROS agnostic. It looks like tf2 core still requires ROS to at least be installed on your system to function (`ros/time.h`, `ros/message_forward.h` etc are included in tf2 headrs), am I wrong about this? I realize a fully independent library might duplicate existing code (creating ros::Time and tf::Time, for instance), but am curious exactly where the coupling between tf2 and ROS is, and how easily tf2 can be used as a standalone library.

tfoote, 2016-05-10:
Yes, we reused those datatypes for efficiency, but it does not use any of the ros communication methods.  To strip it out you just need to copy those datatypes. 

We've iterated a little bit farther and you can find a new version with only the geometry_msgs/TransformStamped datatype as a dependency, due to it being in the public API. https://github.com/ros/geometry2/tree/ros2_alpha

phil0stine, 2016-05-11:
Thanks for the information!

closed 2016-05-11 by phil0stine

fryguy1013, 2017-01-03:
It seems this branch changed to https://github.com/ros/geometry2/tree/ros2 for people coming to this issue.

xudong0401, 2017-10-17:
I can not open the website https://github.com/ros/geometry2/tree/ros2 ,can you update the new website?

mikaelarguedas, 2017-10-17:
The branch use for ros2 development has now been moved to a fork under the ros2 organization: https://github.com/ros2/geometry2

erichlf, 2021-05-20:
This seems to be the real solution: https://github.com/kheaactua/conan-tf2