ros/geometry2 #175: cache.cpp introduces an ODR violation (with tf's cache.cpp)

macmason, 2016-05-11:
TF builds a cache.cpp, which defines (among other things) createExtrapolationException1. For some reason, these are defined in the global namespace.

TF2 defines the same functions, again in the global namespace. As TF depends on TF2, this is an ODR violation, so any call into these functions is undefined behavior.

macmason, 2016-05-11:
I can fix this by using a "namespace tf2 { ... }" rather than a "using namespace tf2" in cache.cpp. Presumably, we'll fallen afoul of some esoteric C++ name lookup rule.

tfoote, 2016-05-11:
Thanks for catching that. I've made it explicit in #176 does that work for you?

macmason, 2016-05-11:
Sure does. One nitpick on the change itself.

closed 2016-05-13 by tfoote