ros/geometry2 #181: More than one static_transform_publisher doesn't work via rosbag

leftys, 2016-07-11:
I tried to run this on Indigo:
```
rosbag record /tf /tf_static
rosrun tf2_ros static_transform_publisher 0 0 0 0 0 0 base_link test
rosrun tf2_ros static_transform_publisher 0 0 0 0 0 0 base_link test2
```
and then 
```rosbag play ...```
wait for a while
```rostopic echo /tf_static```
Only the second tf (base_link -> test2) shows up.

tfoote, 2016-07-11:
Related: https://github.com/ros/ros_comm/issues/146

mkhansen-intel, 2016-11-17:
@tfoote - is there any plan to fix this?

peci1, 2018-08-30:
What about running https://github.com/tradr-project/static_transform_mux during the recording? That would "ensure" that the last published `/tf_static` message would contain all the transforms needed for replay. Moreover, you could even add more static transforms during the replay, if you launch the mux also during replay.

doisyg, 2019-05-14:
I would love this issue to be fixed or @peci1 package to be merged and released.

closed 2025-05-31 by sloretz