ros/geometry2 #183: transform-functions for not-stamped Types in BufferInterface?

cwecht, 2016-08-21:
This issue rather is a question of principle than an issue, but let's start from the beginning. At first, I just wanted to extend tf2_geometry_msgs by doTransform-implementations for not-stamped geometry messages such as Vector3, Point and Pose, because I need this in a project and thought, that this might be interesting for others (as mentioned in #116). Implementations for stamped versions already existed, so I thought, there should be no problem to add the same  implementations for the not-stamped case.

As soon as I tried to add rostests for this cases, I realized that the current Implementation of BufferInterface relies on the existence of a header in the types to be transformed. At this point I want to ask, if there should be version of the overloaded function transform in BufferInterface, which support not-stamped types. I tried this [here](https://github.com/cwecht/geometry2/tree/more_geometry_msgs_conversions) and met the problem of function overloading clashes. I've solved this now by selecting the correct implementation based on the existence of a header (using ros::message_traits and boost::enable_if). This introduces some further complexity, but provides in my opinion a quite good solution. 

So, back to my actual question: is there interest in supporting not-stamped types in BufferInterface? g

tfoote, 2016-08-22:
Adding the doTransform for non-stamped types is ok. It's a lower level interface than was originally designed to be used directly. And is not necessary for the more higher level use case but I don't have a problem with adding more basic datatype support. The doTransform methods are self contained. 

However in general I would recommend using native datatypes and their native representations with their native transformation representations if you just want to do math on transforms. If you have a bullet datatype there is a bullet transform method that can simply be multiplied by a point, pose, or vector to get the result you want. Likewise you can do the same with KDL, eigen, or whatever math library you want to use. So I guess I'd suggest only implementing this for the messages. 

For any implementation without the timestamp they will not be able to be used in the higher level, since you cannot lookup the correct transform without the timestamp. This is why you're having trouble extending the BufferInterface. Conceptually without the timestamp there is a prerequisite missing for looking up the net transform in the buffer. And although you can do fancy things to overload the methods, it's simpler to keep the current interface and construct the Stamped datatype to pass into the methods, since you need all that information anyway.

tfoote, 2017-07-14:
The basic conversions have been added and can work for the core datatypes in #254 

As the Buffer interface needs the extra information to compute the transform explicitly and I think that implementing the doTransform inteface for them has diminishing returns compared to simply encouraging developers to use the native datatype operations. I'm going to close this for now.

closed 2017-07-14 by tfoote