ros/geometry2 #184: ros2 type support implementation introspection_cpp does not match rmw implementation opensplice_static

geoffviola, 2016-08-25:
In the ros2_alpha branch, a std::runtime error gets thrown, when a publisher gets set. The error is as follows:
terminate called after throwing an instance of 'std::runtime_error'
  what(): could not create a publisher: type support implementation 'introspection_cpp' (0x...) does not match rmw implementation 'opensplice_static' (0x...) at .../src/ros2/rmw_opensplice/rmw_opensplice_cpp/src/rmw_publisher.cpp:65, at .../src/ros2/rcl/rcl/src/rcl/publisher.c:81.

This is in ros 2 alpha 7. I tried reordering the include of rclcpp so that it is at the top of the target_link_libraries, but I still see this problem.

geoffviola, 2016-08-25:
This crash may be specific to the static_transform_publisher. It was reproduced on Windows and Linux. It does not crash work in alpha 5.

wjwwood, 2016-08-30:
I think it might be that the `tf_ros` package is not creating a library for each rmw implementation available. Therefore, tf might be built for one implementation while your program or the program you're running is using another.

There was a similar problem with the examples at some point. See: https://github.com/ros2/examples/pull/92/files

geoffviola, 2016-08-31:
Take a look at this gist: https://gist.github.com/geoffviola/77279e5bd582dd2610a1faa16f9d4cd7. I've been trying to make it work, but now I get another exception "Unhandled exception at 0x00007FFF66AC1F28 in static_transform_publisher.exe: Microsoft C++ exception: std::bad_alloc at memory location 0x0000009C012FCEE0." at rclcpp\node.hpp.

I'm testing this on a Windows machine running visual studio 14 2015. I get the same results in the IDE as I get with ament.

tfoote, 2017-07-14:
ros2 stuff should be tracked in https://github.com/ros2/geometry2 for now. I believe that this has been resolved in the mean time.

closed 2017-07-14 by tfoote