ros/geometry2 #185: Shortest path angle calculation

phil0stine, 2016-09-02:
The calculation of the shortestAngle represented by a quaternion, defined [here](https://github.com/ros/geometry2/blob/c0f1905bfca22695a387c59ceffdd8c8a99498af/tf2/include/tf2/LinearMath/Quaternion.h#L235) is the opposite of what it should be, right? If the dot product is less than zero, the calculation should be made on -q.w. Currently this happens if the dot is greater than zero.

tfoote, 2016-09-07:
This is a fork of the Bullet LinearMath library. https://github.com/bulletphysics/bullet3/blob/master/src/LinearMath/btQuaternion.h#L428-L437

It looks like they were patched here: https://github.com/bulletphysics/bullet3/commit/e80bafdf6639429193ca5601d195c54f7793a416

A PR with the same change would be appreciated.

tfoote, 2016-09-08:
fixed in #187

closed 2016-09-08 by tfoote

moucrob, 2018-11-06:
Hi, given a frame1 which is the rotation by a quaternion q1 from a frame0, and a frame2 which is another rotation of frame0 by q2, what is the **shortest {angle**, axis} that leads frame1 to become frame2 ?

I would use : 
```
tf2Scalar angleShortestPath(const Quaternion& q) const 
	{
		tf2Scalar s = tf2Sqrt(length2() * q.length2());
		tf2Assert(s != tf2Scalar(0.0));
		if (dot(q) < 0) // Take care of long angle case see http://en.wikipedia.org/wiki/Slerp
			return tf2Acos(dot(-q) / s) * tf2Scalar(2.0);
		else 
			return tf2Acos(dot(q) / s) * tf2Scalar(2.0);
}
```
from https://github.com/ros/geometry2/blob/melodic-devel/tf2/include/tf2/LinearMath/Quaternion.h line 218 but I tried some simple scenarios in a matlab script which is : 
**testShortestAngleBetweenQuaternions.m**
```
%% 2 successives rotations of something around fixed z
clear all ; clc
z = [0;0;1] ; toNormalize=1;
angle1 = pi/4 ; angle2 = pi/2 ; angle3 = 8*pi/6;
angle3re=angle3; angle3 = from02piToMinusPiPi(angle3);
fprintf('Starting from the horizontal right line, \n')
fprintf(['angle1 = ',num2str(rad2deg(angle1)),', '])
fprintf(['angle2 = ',num2str(rad2deg(angle2)),', '])
fprintf(['angle3 = ',num2str(rad2deg(angle3re)),'\n'])
fprintf(['angle3 after reshape = ',num2str(rad2deg(angle3)),'\n'])
disp(' ')

%% visual plot of the angles
clf
angleBetween12 = rad2deg(linspace(angle1,angle2))/180*pi;
patch([0 cos(angleBetween12) 0], [0 sin(angleBetween12) 0], 'r')
grid on ; grid minor ; xlim([-1 1]) ; ylim([-1 1])
set(gca,'DataAspectRatio',[1,1,1])
angleBetween13 = rad2deg(linspace(angle1,angle3))/180*pi;
patch([0 cos(angleBetween13) 0], [0 sin(angleBetween13) 0], 'g')

%% Call to the functions (defined below)
h1 = quat(angle1,z,toNormalize);
h2 = quat(angle2,z,toNormalize);
h3 = quat(angle3,z,toNormalize);
% all with respect to global frame

%% Outputs
fprintf('1st implementation (rb). Shortest angle \n ')
fprintf(['from after-h1-rotation configuration \n to ' ...
      'after-h2-rotation configuration = ', num2str(diffQuat1(h1,h2)*180/pi), '°. \n'])
fprintf(['(expected : ', num2str(rad2deg(shortestDiffBetween2Angles(angle2-angle1))), '°.) \n'])
fprintf(['shortest angle \n from after-h1-rotation configuration \n to ' ...
      'after-h3-rotation configuration = ', num2str(diffQuat1(h1,h3)*180/pi), '°. \n'])
fprintf(['(expected : ', num2str(rad2deg(shortestDiffBetween2Angles(angle3-angle1))), '°.) \n'])
disp('which is wrong.')
disp(' ')

fprintf('2nd implementation. Shortest angle \n ')
fprintf(['from after-h1-rotation configuration \n to ' ...
      'after-h2-rotation configuration = ', num2str(diffQuat2(h1,h2,0)*180/pi), '°. \n'])
fprintf(['(expected : ', num2str(rad2deg(shortestDiffBetween2Angles(angle2-angle1))), '°.) \n'])
fprintf(['shortest angle \n from after-h1-rotation configuration \n to ' ...
      'after-h3-rotation configuration = ', num2str(diffQuat2(h1,h3,0)*180/pi), '°. \n'])
fprintf(['(expected : ', num2str(rad2deg(shortestDiffBetween2Angles(angle3-angle1))), '°.) \n'])
disp('which is NOT ALWAYS true, unless the input angles lie into [-pi;pi].')
disp('for q1=(45°,z) and q2= either (90°,z) or (-45°,z) it works,')
disp('but for q1=(45°,z) and q2=-4pi/3 it doesnot')
disp(' ')

fprintf('3rd implementation (current ros). Shortest angle \n ')
fprintf(['from after-h1-rotation configuration \n to ' ...
      'after-h2-rotation configuration = ', num2str(diffQuat3(h1,h2,0)*180/pi), '°. \n'])
[... 79 lines cut ...]
 to after-h2-rotation configuration = 45°. 
(expected : 45°.) 
shortest angle 
 from after-h1-rotation configuration 
 to after-h3-rotation configuration = 165°. 
(expected : -165°.) 
which is NOT ALWAYS true, unless the input angles lie into [-pi;pi].
for q1=(45°,z) and q2= either (90°,z) or (-45°,z) it works,
but for q1=(45°,z) and q2=-4pi/3 it doesnot

> 3rd implementation (current ros). Shortest angle 
 from after-h1-rotation configuration 
 to after-h2-rotation configuration = 135°. 
(expected : 45°.) 
shortest angle 
 from after-h1-rotation configuration 
 to after-h3-rotation configuration = 15°. 
(expected : -165°.) 
which is NOT ALWAYS true.

and I manage to make the current [wikipedia formula](https://en.wikipedia.org/wiki/Quaternions_and_spatial_rotation#Recovering_the_axis-angle_representation) work only if I assume that the angle in quaternions belong necessarily to [-pi, pi].

**Where could I look at to figure out if this is how they are build in ROS please? Regards**

@dtecta @erwincoumans @tfoote @phil0stine @AndyZe

dtecta, 2018-12-13:
Although my name is mentioned in the source code, I am not the author of the source code. I see that the code may be derived from my quaternion code either in Blender or SOLID3, but I don't know the details of this code. 

What I do see is that the expression "tf2Acos(dot(q) / s) * tf2Scalar(2.0)"  makes sense if "dot(q)" is regarded to be the scalar part of q1 * conjugate(q2). Thus, the scalar part of  (v1, s1) * (-v2, s2) is indeed v1.dot(v2) + s1 * s2 = q1.dot(q2). Could it be that conjugate(q2) is used as argument of "angleShortestPath" rather than simply q2?

Gino