ros/geometry2 #190: setEuler bad parameter names

geoffviola, 2016-10-07:
When I use setEuler in https://github.com/ros/geometry2/blob/indigo-devel/tf2/include/tf2/LinearMath/Quaternion.h#L73. It appears to be pitch, roll, yaw and not yaw, pitch, roll. 

I'm setting a TF's rotation to this quaternion. The RPY to check against is from the "rosrun tf tf_echo" command. Also, rviz seems to match what tf_echo prints out.

tfoote, 2017-01-09:
Can you provide a simple example/unit test showing that these are out of order? This is an internal fork of bullet's linear math: https://github.com/bulletphysics/bullet3/blob/master/src/LinearMath/btQuaternion.h#L120-L139

Note that it's strongly recommended to use bullet's LinearMath library directly and not the internal tf2 version.

tfoote, 2017-07-10:
Closing due to no follow up. Also if it's strictly reversed it may be an issue of perspective, whether it's the transform to the new frame, or the transform of an object into the target frame.

closed 2017-07-10 by tfoote

corot, 2019-05-10:
Actually this is a legitimate bug: using setEuler with the documented parameter order doesn't work as expected. Here ROS nav people "fixed" [the issue using setRPY](https://github.com/ros-planning/navigation/issues/805) instead, but the problem remains on this repo.

To my understanding, [setRPY](https://github.com/ros/geometry2/blob/04318f537e5ca86b9a85c8c6971802b2c6faf049/tf2/include/tf2/LinearMath/Quaternion.h#L95) and [setEuler](https://github.com/ros/geometry2/blob/04318f537e5ca86b9a85c8c6971802b2c6faf049/tf2/include/tf2/LinearMath/Quaternion.h#L75) code should be the same, just changing the order of the parameters (RPY vs. YPR). However, the matrix calculation differ:

```
		setValue(cosRoll * sinPitch * cosYaw + sinRoll * cosPitch * sinYaw,
			cosRoll * cosPitch * sinYaw - sinRoll * sinPitch * cosYaw,
			sinRoll * cosPitch * cosYaw - cosRoll * sinPitch * sinYaw,
			cosRoll * cosPitch * cosYaw + sinRoll * sinPitch * sinYaw);

		setValue(sinRoll * cosPitch * cosYaw - cosRoll * sinPitch * sinYaw, //x
                         cosRoll * sinPitch * cosYaw + sinRoll * cosPitch * sinYaw, //y
                         cosRoll * cosPitch * sinYaw - sinRoll * sinPitch * cosYaw, //z
                         cosRoll * cosPitch * cosYaw + sinRoll * sinPitch * sinYaw); //formerly yzx
```
What leads to the behavior @geoffviola describes