ros/geometry2 #222: Include transformations.py in tf2?

stwirth, 2017-04-05:
The [tf->tf2 migration page](http://wiki.ros.org/tf2/Migration) says "Since ROS Hydro, tf has been "deprecated" in favor of tf2". tf includes the tremendously useful transformations.py module and we therefore are often including both tf and tf2_ros in our python nodes. To make the transition cleaner, I think including `transformations.py` in tf2 or tf2_ros would be great.

tfoote, 2017-04-05:
transformations.py is effectively an external python linear math library. Bundling it into tf2 doesn't really make sense. We've been working hard to strip down dependencies and reduce maintenance of these core tools. For example external linear math libraries bullet, eigen and kdl are no longer core dependencies.

I'd rather an effort to package is separately. it's also evolved into transofrms3d and there was an initial attempt to make this available in kinetic here: https://github.com/ros/rosdistro/pull/12437 I think with a little bit of build finessing we could make it work that way: https://github.com/ros/rosdistro/pull/12437

stwirth, 2017-04-05:
I'd be totally fine with that! Closing this then.

closed 2017-04-05 by stwirth