ros/geometry2 #224: tf2_geometry_msgs C++ interface is not consistent

victl, 2017-04-09:
Hi,
I've been using tf2 for some days. One thing I could not understand is why the package `tf2_geometry_msgs` has inconsistent C++ interface.

An example is `to_msg()` and `from_msg()`. The later returns `void` and values were passed by reference. While the former returns the desired message value directly.

This inconsistency makes me feels `tf2` is far less mature than tf. Should I choose tf2 in favor  of tf?

tfoote, 2017-04-13:
Thanks for catching the inconsistency. I've added unit tests that provide coverage and am debugging it in this branch https://github.com/ros/geometry2/tree/pAIgn10-indigo-devel in reviewing and extending #202

tfoote, 2017-07-14:
Sorry I misread this last time I reviewed it.

What i was referring to is the inconsistency is between Vector3 and Point  as well as between Transform and Pose return types specifically. These prototypes are explicitly different because they share the same underlying representation and the limitations of template type deduction based on return type.

The toMsg fromMsg prototypes were designed to be used under the hood to support the needs of convert.h. We could expand them out to be more feature complete, however in the current state we generally expect users to use the `convert()` method for which the current API is adequate.

At the high level you should use tf2. If you're using tf, you're actually using tf2 as tf is now implemented as a backwards compatibility layer on top of tf2. I'm going to close this as I don't currently have plans to extend the toMsg/fromMsg API and recommend just using the convert method for users. It's much more compact and easier to remember. And the tf2 API is templated to automatically invoke it too.

closed 2017-07-14 by tfoote