ros/geometry2 #228: tf2_urdf

vincentrou, 2017-05-02:
Could it be interesting to have a package tf2_urdf which can convert [urdf types](https://github.com/ros/urdfdom_headers/tree/master/urdf_model/include/urdf_model) ?

This is linked to this new package : https://github.com/ros-controls/urdf_utilities/pull/1

tfoote, 2017-05-02:
That could be an interesting thing to do. I'd suggest that it be housed in a different repository so as not to bring in a urdf dependency into the geometry2 repo.

And I'd suggest it might make sense to go side by side with your new repository urdf_utilities. However I'd suggest the catch all utilities name be dropped for a more specific name with specific scope: http://wiki.ros.org/ROS/Patterns/Conventions

tfoote, 2017-07-10:
I'm going to close this ticket here as it's not something to track in this repository. But I do encourage the creation of tf2_urdf to happen.

closed 2017-07-10 by tfoote