ros/geometry2 #236: Feature Request (Python): Get Parent of a given Frame

Sanic, 2017-06-05:
Is there a way to get the parent of a frame in a TF2Buffer when using Python?
tf2buffer.all_frames_as_string() provides this data in a human readable form, but i would like access the parent of a given frame name to analyze the hierarchy of the buffered frames.

tfoote, 2017-06-05:
It would be possible to add this feature to the python API but it would actually be just doing exactly what you are doing above as the raw tree is not available in the python implementation, everything passes through to the C++ API. As such I'm not sure how much it's worth hiding that implementation since as you've shown it can be done in 2 lines of code. 

And adding a full layer to support passing the datastructure through is adding a lot of complexity for a use case that is already supported.

Sanic, 2017-06-06:
Hi @tfoote 
Thanks for your quick reply.
I totally agree that it's not a viable option to pass through the whole datastructure just to realize this use case.
However, from a code duplication point of view i was thinking that it may be reasonable to add a method called all_frames_as_dict to the Python API to avoid calling YAML every time in my caller code. This is a bit of cherry picking, but should lead to cleaner/shorter code on the caller side.

tfoote, 2017-06-07:
Yeah, that would be a reasonable extension if you'd like to submit a PR for that.

closed 2025-05-31 by sloretz