ros/geometry2 #240: lookup_transform fails only first time

furushchev, 2017-06-16:
I'm using TF2 to resolve transformation in my node, but `lookup_transform` always fails first time on each nodes.

I first spawn `tf2_buffer_server` like below:

```xml
<launch>
  <node pkg="tf2_ros" type="buffer_server" name="tf2_buffer_server">
    <param name="buffer_size" value="120.0"/>
  </node>
</launch>
```

Then each nodes use TF2 interface like below:

```python
import traceback
 import rospy
 import tf2_ros
 from geometry_msgs.msg import PointStamped

 rospy.init_node("aaa")
 tfl = tf2_ros.BufferClient("tf2_buffer_server")
 p = PointStamped()
 p.header.frame_id = "base_link"
 p.header.stamp = rospy.Time.now() - rospy.Duration(10)
 p.point.x = 1.0

 try:
     t = tfl.lookup_transform("odom", p.header.frame_id,
                              p.header.stamp, rospy.Duration(0))
     rospy.loginfo("transform 1: %s" % t)
 except Exception as e:
     print str(e)
     traceback.print_exc()

 try:
     t = tfl.lookup_transform("odom", p.header.frame_id,
                              p.header.stamp, rospy.Duration(0))
     rospy.loginfo("transform 2: %s" % t)
 except Exception as e:
     print str(e)
     traceback.print_exc()

 try:
     t = tfl.lookup_transform("odom", p.header.frame_id,
                              p.header.stamp, rospy.Duration(0))
     rospy.loginfo("transform 3: %s" % t)
 except Exception as e:
     print str(e)
     traceback.print_exc()

 try:
     t = tfl.lookup_transform(p.header.frame_id, "odom", p.header.stamp)
     rospy.loginfo("transform 4: %s" % t)
 except Exception as e:
     print str(e)
     traceback.print_exc()
```

Output is here:

```bash
$ ./test_tf.py 
The LookupTransform goal sent to the BufferServer did not come back in the specified time. Something is likely wrong with the server
Traceback (most recent call last):
  File "./test_tf.py", line 20, in <module>
    p.header.stamp, rospy.Duration(0))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/tf2_ros/buffer_client.py", line 95, in lookup_transform
    return self.__process_goal(goal)
  File "/opt/ros/indigo/lib/python2.7/dist-packages/tf2_ros/buffer_client.py", line 185, in __process_goal
    raise tf2.TimeoutException("The LookupTransform goal sent to the BufferServer did not come back in the specified time. Something is likely wrong with the server")
TimeoutException: The LookupTransform goal sent to the BufferServer did not come back in the specified time. Something is likely wrong with the server
[INFO] [WallTime: 1497607676.264143] transform 2: header: 
  seq: 0
  stamp: 
    secs: 1497607664
    nsecs:  44776916
  frame_id: odom
child_frame_id: base_link
transform: 
  translation: 
    x: 0.123658795259
    y: 0.121590551562
    z: 0.0
  rotation: 
    x: 0.0
    y: 0.0
    z: -0.762227715139
    w: 0.647308975895
[INFO] [WallTime: 1497607676.366201] transform 3: header: 
  seq: 0
  stamp: 
    secs: 1497607664
    nsecs:  44776916
  frame_id: odom
child_frame_id: base_link
transform: 
  translation: 
    x: 0.123658795259
    y: 0.121590551562
    z: 0.0
  rotation: 
    x: 0.0
    y: 0.0
    z: -0.762227715139
    w: 0.647308975895
[INFO] [WallTime: 1497607676.471199] transform 4: header: 
  seq: 0
  stamp: 
    secs: 1497607664
    nsecs:  44776916
  frame_id: base_link
child_frame_id: odom
transform: 
  translation: 
    x: 0.140015309403
    y: -0.102330215512
    z: 0.0
  rotation: 
    x: -0.0
    y: -0.0
    z: 0.762227715139
    w: 0.647308975895
```

TF itself is published in high frequency

```bash
$ rosrun tf tf_monitor base_link odom
Waiting for transform chain to become available between base_link and odom 

RESULTS: for base_link to odom
Chain is: base_link -> base_footprint -> odom
Net delay     avg = 0.00654071: max = 0.066185

Frames:
Frame: base_footprint published by unknown_publisher Average Delay: 0.0167895 Max Delay: 0.029394
Frame: base_link published by unknown_publisher Average Delay: -0.475396 Max Delay: 0
Frame: odom published by unknown_publisher Average Delay: 0.00339817 Max Delay: 0.0166253

All Broadcasters:
Node: unknown_publisher 342.304 Hz, Average Delay: -0.127553 Max Delay: 0.0418373
```

furushchev, 2017-06-16:
I found that if `rospy.sleep(1)` is inserted after instantiation of `tf2_ros.BufferClient`, all transformation can be resolved.
I guess it is necessary to wait enough for advertising goal topic of tf2 actionlib from client to t`tf2_buffer_server`.

closed 2017-06-16 by furushchev

tfoote, 2017-06-16:
There's some setup time to establish connections between the BufferServer and BufferClient that's likely what is causing your first query to timeout. If you send the first request before the topic is setup it will never reach the far end and you'll incur the full timeout waiting.

I'd recommend using waitForServer instead of sleep as it will return as soon as the connection is established instead of waiting the full time until the timeout.

furushchev, 2017-06-17:
@tfoote Thanks for advice! Now all frames can be solved!