ros/geometry2 #242: Cmake error

hjjayakrishnan, 2017-07-03:
I just  did a `sudo apt-get remove ros-indigo-tf2*`, and I'm trying to build it from source. This is because I wanted to relax the **TF_DENORMALIZED_QUATERNION** tolerances that were tightened recently. I think the tolerances were relaxed after an issue was raised but the pre-built binaries from sudo apt-get still retains the tightened tolerances.

```
Base path: /home/correlllab/tf2_ws
Source space: /home/correlllab/tf2_ws/src
Build space: /home/correlllab/tf2_ws/build
Devel space: /home/correlllab/tf2_ws/devel
Install space: /home/correlllab/tf2_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/correlllab/tf2_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/correlllab/tf2_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/correlllab/tf2_ws/devel;/home/correlllab/ros/jaco_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/correlllab/tf2_ws/devel;/home/correlllab/ros/jaco_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/correlllab/tf2_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.19
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 13 packages in topological order:
-- ~~  - geometry2 (metapackage)
-- ~~  - geometry_experimental (metapackage)
-- ~~  - tf2_msgs
-- ~~  - tf2
-- ~~  - tf2_bullet
-- ~~  - tf2_eigen
-- ~~  - tf2_py
-- ~~  - tf2_ros
-- ~~  - tf2_geometry_msgs
-- ~~  - tf2_kdl
-- ~~  - test_tf2
-- ~~  - tf2_sensor_msgs
-- ~~  - tf2_tools
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin metapackage: 'geometry2'
-- ==> add_subdirectory(geometry2/geometry2)
-- +++ processing catkin metapackage: 'geometry_experimental'
-- ==> add_subdirectory(geometry2/geometry_experimental)
-- +++ processing catkin package: 'tf2_msgs'
-- ==> add_subdirectory(geometry2/tf2_msgs)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   thread
-- Generating .msg files for action tf2_msgs/LookupTransform /home/correlllab/tf2_ws/src/geometry2/tf2_msgs/action/LookupTransform.action
-- tf2_msgs: 9 messages, 1 services
-- +++ processing catkin package: 'tf2'
-- ==> add_subdirectory(geometry2/tf2)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   signals
--   system
--   thread
-- +++ processing catkin package: 'tf2_bullet'
-- ==> add_subdirectory(geometry2/tf2_bullet)
-- checking for module 'bullet'
--   found bullet, version 2.81
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'tf2_eigen'
-- ==> add_subdirectory(geometry2/tf2_eigen)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'tf2_py'
-- ==> add_subdirectory(geometry2/tf2_py)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'tf2_ros'
-- ==> add_subdirectory(geometry2/tf2_ros)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   thread
-- +++ processing catkin package: 'tf2_geometry_msgs'
-- ==> add_subdirectory(geometry2/tf2_geometry_msgs)
-- Using these message generators: gencpp;genlisp;genpy
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   thread
-- +++ processing catkin package: 'tf2_kdl'
-- ==> add_subdirectory(geometry2/tf2_kdl)
-- Using these message generators: gencpp;genlisp;genpy
-- +++ processing catkin package: 'test_tf2'
-- ==> add_subdirectory(geometry2/test_tf2)
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "tf" with any of
  the following names:

    tfConfig.cmake
    tf-config.cmake

  Add the installation prefix of "tf" to CMAKE_PREFIX_PATH or set "tf_DIR" to
  a directory containing one of the above files.  If "tf" provides a separate
  development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  geometry2/test_tf2/CMakeLists.txt:10 (find_package)

-- Could not find the required component 'tf'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "tf" with any of
  the following names:

    tfConfig.cmake
    tf-config.cmake

  Add the installation prefix of "tf" to CMAKE_PREFIX_PATH or set "tf_DIR" to
  a directory containing one of the above files.  If "tf" provides a separate
  development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  geometry2/test_tf2/CMakeLists.txt:10 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/correlllab/tf2_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/correlllab/tf2_ws/build/CMakeFiles/CMakeError.log".
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

```

hjjayakrishnan, 2017-07-03:
Fixed the issue. It was a tf dependency issue. Cloning the tf repository and building fixed it.

closed 2017-07-03 by hjjayakrishnan

kaanoguzhan, 2020-05-27:
@hjjayakrishnan  It would be nice if you can also mention "tf repository"