ros/geometry2 #243: Suggestion to change tf2 tutorial node names

jwhendy, 2017-07-05:
I'm going through the tutorials for `tf2` and noticed that the node names are different from `tf`. I saved my `turtle_tf2_broadcaster.py` as `turtle_tf2_listener.py` to re-use the imports and was looking for other changes when I spotted the `rospy.init_node()` call and had to double check myself:

[Writing a tf2 broadcaster](http://wiki.ros.org/tf2/Tutorials/Writing%20a%20tf2%20broadcaster%20%28Python%29)
```
if __name__ == '__main__':
    rospy.init_node('tf_turtle')
```

[Writing a tf2 listener](http://wiki.ros.org/tf2/Tutorials/Writing%20a%20tf2%20listener%20%28Python%29)
```
if __name__ == '__main__':
    rospy.init_node('tf2_turtle')
```

The [original tf tutorials](http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20broadcaster%20%28Python%29) use more explicit node names like this:
```
if __name__ == '__main__':
    rospy.init_node('turtle_tf_broadcaster')
```

I'm pretty new to ROS, so perhaps when one is using a launch file it doesn't really matter. I noticed that following along with the tutorials and using `roslaunch` yields helpful node names like `/turtle1_tf2_broadcaster` since they're defined in the launch file. I'm still going to guess it's not great practice for new people (like me!) to use generic sounding node names. Running `roscore` and then the broadcaster explicitly gives this:

```
$ rosrun learning_tf2 turtle_tf2_broadcaster.py _turtle:='turtle1'
$ rosnode list
/rosout
/tf_turtle
```
It seems easily avoidable to not have to wonder whether `tf_turtle` or `tf2_turtle` is the listener or broadcaster node.

I can change this on the wiki if you agree. I'd propose just mimicking the style with `tf`, but replacing that with `tf2` for the node names.

tfoote, 2017-07-05:
Homogenizing those would be greatly appreciated! If you could validate that they match the repo too that would be even better: https://github.com/ros/geometry_tutorials/tree/indigo-devel/turtle_tf2/nodes

jwhendy, 2017-07-05:
Sounds good. Give me a couple days I can do the wiki edits and PRs for the files you pointed to.

jwhendy, 2017-07-05:
Well, why wait when that's more fun that "work"? Here's the current status:

```
~/installed/build/geometry_tutorials$ grep -r init_node *
turtle_tf/nodes/dynamic_tf_broadcaster.py:    rospy.init_node('my_tf_broadcaster')
turtle_tf/nodes/dynamic_tf_broadcaster.py:    rospy.init_node('my_tf_broadcaster')
turtle_tf/nodes/fixed_tf_broadcaster.py:    rospy.init_node('my_tf_broadcaster')
turtle_tf/nodes/turtle_tf_broadcaster.py:    rospy.init_node('tf_turtle_broadcaster')
turtle_tf/nodes/turtle_tf_listener.py:    rospy.init_node('tf_turtle_listener')
turtle_tf/nodes/turtle_tf_listener_wait.py:    rospy.init_node('tf_turtle_listener_wait')
turtle_tf/nodes/turtle_tf_message_broadcaster.py:    rospy.init_node('tf_turtle_stamped_msg_publisher')
turtle_tf2/nodes/dynamic_tf2_broadcaster.py:    rospy.init_node('my_tf2_broadcaster')
turtle_tf2/nodes/dynamic_tf2_broadcaster.py:    rospy.init_node('my_tf2_broadcaster')
turtle_tf2/nodes/fixed_tf2_broadcaster.py:    rospy.init_node('my_tf2_broadcaster')
turtle_tf2/nodes/turtle_tf2_broadcaster.py:    rospy.init_node('tf2_turtle_broadcaster')
turtle_tf2/nodes/turtle_tf2_listener.py:    rospy.init_node('tf2_turtle_listener')
turtle_tf2/nodes/turtle_tf2_listener_wait.py:    rospy.init_node('tf2_turtle_listener_wait')
turtle_tf2/nodes/turtle_tf2_message_broadcaster.py:    rospy.init_node('tf2_turtle_stamped_msg_publisher')
```

Per both `tf` and `tf2` tutorials, there are three instances of `my_tf2_broadcaster`. Do those need to be changed? Otherwise I'd say this is fixed (changed on wiki too).

tfoote, 2017-07-05:
That looks good. The dynamic example could add 'dynamic' into the name for clarity. And the 2 instances inside look like the top one is actually never used, and could be at least commented out, if not removed since it's a duplicate of the fixed tf broadcaster.

jwhendy, 2017-07-05:
Rethinking this... any reason not to just use the name of the file, or if the convention is desired, flip the `tf[2]` and `turtle`? At present they go pretty well with `turtle_tf_blah.py` and `init_node('tf_turtle_blah')`.

If there's other wisdom about file names vs. node names, perhaps it could be leveraged here? It wouldn't seem to harm anything to just go more specific/clear vs. not as long as we're at it :)

tfoote, 2017-07-05:
Matching the filename seems good to allow unfamiliar users to associate the two. The init names are mostly legacy and since they're not a primary part of the tutorial they likely didn't get touched in refactoring.

jwhendy, 2017-07-05:
Wiki needs another go through since we changed strategies to match the file name, but just submitted the PR.

jwhendy, 2017-07-06:
Wiki is updated with matching node names now.

closed 2017-07-06 by jwhendy

tfoote, 2017-07-06:
@jwhendy Thanks!