ros/geometry2 #245: issue with past extrapolation in time travel tf2 tutorial

jwhendy, 2017-07-05:
Sigh. I wrote all of the below about [Learning about tf2 and time (Python)](http://wiki.ros.org/tf2/Tutorials/tf2%20and%20time%20%28Python%29) and then it failed with an error after restoring it back to `rospy.Time()`. Huh.

Well, I guess if one leaves the `raise` in the `except():` block, the exceptions get raised as things startup and we wait for `tf2` to kick in (half guessing)?

Could someone kindly confirm that this:

>Also add to be able to see the exception reraise it
```
raise
rate.sleep() 
```

should be followed somewhere with something like this:

"After implementing the `rospy.Duration()` parameter to wait for the transform to be available, **also remove the `raise` line** from the exceptions so that things continue to run while the transforms are initiated."

---

**Original issue for posterity's sake**

I've been following along with the `tf2` tutorials and am hung up on [Learning about tf2 and time (Python)](http://wiki.ros.org/tf2/Tutorials/tf2%20and%20time%20%28Python%29). Interestingly, I got hung up on the exact same tutorial in the `tf` series, asking about it on [ROS Answers](http://answers.ros.org/question/265272/tf-transform-tutorial-issue-past-extrapolation/).

To reproduce, I've pushed my code to github.

```
cd ~/catkin_ws/src
git clone https://github.com/jwhendy/learning_tf2.git
cd ~/catkin_ws
catkin build

roslaunch learning_tf2 start_demo.launch
```
It's puzzling to me that *both* attempts to follow along with the tutorials would fail.

Here's the session output:
```

$ roslaunch learning_tf2 start_demo.launch 
... logging to /home/jwhendy/.ros/log/d749eb34-61ce-11e7-a203-cc3d82b9eef7/roslaunch-roboxenial-26085.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://roboxenial:36832/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /turtle1_tf2_broadcaster/turtle: turtle1
 * /turtle2_tf2_broadcaster/turtle: turtle2

NODES
  /
    broadcaster_fixed (learning_tf2/fixed_tf2_broadcaster.py)
    sim (turtlesim/turtlesim_node)
    teleop (turtlesim/turtle_teleop_key)
    turtle1_tf2_broadcaster (learning_tf2/turtle_tf2_broadcaster.py)
    turtle2_tf2_broadcaster (learning_tf2/turtle_tf2_broadcaster.py)
    turtle_tf2_listener (learning_tf2/turtle_tf2_listener.py)

auto-starting new master
process[master]: started with pid [26096]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d749eb34-61ce-11e7-a203-cc3d82b9eef7
process[rosout-1]: started with pid [26109]
started core service [/rosout]
process[sim-2]: started with pid [26116]
process[teleop-3]: started with pid [26127]
process[turtle1_tf2_broadcaster-4]: started with pid [26133]
Reading from keyboard
---------------------------
Use arrow keys to move the turtle.
process[turtle2_tf2_broadcaster-5]: started with pid [26142]
process[turtle_tf2_listener-6]: started with pid [26143]
process[broadcaster_fixed-7]: started with pid [26144]
Traceback (most recent call last):
  File "/home/jwhendy/catkin_ws/src/learning_tf2/src/turtle_tf2_listener.py", line 32, in <module>
    rospy.Duration(3.0))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/tf2_ros/buffer.py", line 87, in lookup_transform
    return self.lookup_transform_core(target_frame, source_frame, time)
tf2.ExtrapolationException: Lookup would require extrapolation into the past.  Requested time 1499292667.176356077 but the earliest data is at time 1499292667.311480999, when looking up transform from frame [carrot1] to frame [turtle2]
[turtle_tf2_listener-6] process has died [pid 26143, exit code 1, cmd /home/jwhendy/catkin_ws/src/learning_tf2/src/turtle_tf2_listener.py __name:=turtle_tf2_listener __log:=/home/jwhendy/.ros/log/d749eb34-61ce-11e7-a203-cc3d82b9eef7/turtle_tf2_listener-6.log].
log file: /home/jwhendy/.ros/log/d749eb34-61ce-11e7-a203-cc3d82b9eef7/turtle_tf2_listener-6*.log
[broadcaster_fixed-7] killing on exit
[turtle2_tf2_broadcaster-5] killing on exit
[turtle1_tf2_broadcaster-4] killing on exit
[teleop-3] killing on exit
[sim-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

tfoote, 2017-07-06:
Yes, removing the exception is necessary to continue.

jwhendy, 2017-07-06:
Wiki edited (feel free to double check/modify).

closed 2017-07-06 by jwhendy