ros/geometry2 #247: Unify method for learning about tf frames (class or not)

jwhendy, 2017-07-06:
I just finished going through both the `tf` and `tf2` tutorials and was puzzled by the tutorial on [Adding a frame (Python)](http://wiki.ros.org/tf2/Tutorials/Adding%20a%20frame%20%28Python%29) using `tf2`. The first method for applying a fixed frame is:

```
class FixedTFBroadcaster:

    def __init__(self):
        self.pub_tf = rospy.Publisher("/tf", tf2_msgs.msg.TFMessage, queue_size=1)

        while not rospy.is_shutdown():
            # Run this loop at about 10Hz
            rospy.sleep(0.1)

            # define transform
            
            tfm = tf2_msgs.msg.TFMessage([t])
            self.pub_tf.publish(tfm)

if __name__ == '__main__':
    rospy.init_node('fixed_tf2_broadcaster')
    tfb = FixedTFBroadcaster()

    rospy.spin()
```

Later, we get this odd kludge of what appears to be the "normal" (non-class) way of doing the fixed frame *and* a way to do a dynamic frame.

```
if __name__ == '__ STATIC main__':
    rospy.init_node('fixed_tf2_broadcaster')
    br = tf2_ros.TransformBroadcaster()
    t = geometry_msgs.msg.TransformStamped()
    # define header/child_frame_id/transform
    
    rate = rospy.Rate(10.0)
    while not rospy.is_shutdown():
        t.header.stamp = rospy.Time.now()
        br.sendTransform(t)
        rate.sleep()

if __name__ == '__main__':
    rospy.init_node('dynamic_tf2_broadcaster')
    br = tf2_ros.TransformBroadcaster()
    t = geometry_msgs.msg.TransformStamped()
    # define t header/child_frame_id
 
    rate = rospy.Rate(10.0)
    while not rospy.is_shutdown():

        # define transform

        br.sendTransform(t)
        rate.sleep()
```

Being new to ROS and also relatively new to `python`:

- it was confusing seeing two ways of defining a fixed frame with no explanation of why one and then the other
- I wondered if `__ STATIC main__` was some special `python` keyword and spent time googling for it and asking on [ROS Answers](http://answers.ros.org/question/265532/if-__name__-__-static-main__-in-tf2-tutorials/). My current guess is that it's not and was just some way to make the code not run, as `__name__` will simply never be equivalent
- it doesn't match the `tf` tutorials, which never introduce creating a class. This isn't super critical if classes are preferred, but if they *aren't* it seems to arbitrarily depart from the flow and introduce something else to potentially confuse

Any insight into why this is like this? For now, I've commented out the top part as "for reference"/"another way to do a fixed frame", but unless there's an advantage to the class method I'd vote for replacing the class-based method (Sec 3.1) with the redundant "normal way" code in Sec 6.

jwhendy, 2017-07-06:
Created a branch for this if it's desirable. The proposal would be that these are the new files for fixed and dynamic frame broadcasting, which would then be put into the wiki as well:

- [fixed_tf2_broadcaster.py](https://github.com/jwhendy/geometry_tutorials/blob/unify-tf2-frame-broadcasters/turtle_tf2/nodes/fixed_tf2_broadcaster.py)
- [dynamic_tf2_broadcaster.py](https://github.com/jwhendy/geometry_tutorials/blob/unify-tf2-frame-broadcasters/turtle_tf2/nodes/dynamic_tf2_broadcaster.py)

The main question is whether classes are desirable.

My interpretation having just learned was that the class-based way looked like a lot of extra work; it appeared to be manually defining the same transform, but then going through an extra step of stuffing it into a `tf2_ros.msg.TFMessage` and using a publisher to push it to the `/tf` topic.

It appears that the `TransformBroadcaster` can do all of this for us with the raw transform?

tfoote, 2017-07-15:
These examples are quite old and have a lot of history. Some of them even predate the tools.

For the class based vs non-class based. We were using different ways to show different approaches at some point. Some of these examples used to stand as the way to learn how to use ROS APIs in general.

The simplified versions look great. I'll note that the name in the fixed example says dynamic though.

jwhendy, 2017-07-22:
Sorry for the delay in getting back to this. Is it beneficial to have the class-based method as it's own tutorial to show this way? Personally (and I keep reiterating that I'm super new), I found it confusing as it wasn't acknowledged that the class based method was anything special. Thus, at first I wondered if that was just some new, de facto way to accomplish the task. Then in the next tutorial it was back to the way earlier tutorials worked.

So, my vote would be to keep the fixed/dynamic as "normal" and maybe introduce that classes can be used (and advantages/reasons for doing this) as a separate thing.

It probably doesn't help that I'm relatively new to `python` as well, so learning `tf` *and* having littler familiarity with classes probably doesn't help :)

Thanks for the catch on the dynamic node name. I goofed in my copy/pasting!

tfoote, 2017-07-27:
No problem. If you could submit those as PRs that would be great!

jwhendy, 2017-08-03:
Fixed the node name goof and submitted PR.

130s, 2019-01-19:
https://github.com/ros/geometry_tutorials/pull/27 is merged already so this one should be closed?

jwhendy, 2019-01-19:
Been so long I have no idea :) I'm fine with that and see it's been merged.

closed 2019-01-19 by jwhendy