ros/geometry2 #250: No python API appears to exist for tf2_geometry_msgs

jwhendy, 2017-07-07:
In trying to learn about converting a pose, I stumbled on [this answer](http://answers.ros.org/question/222306/transform-a-pose-to-another-frame-with-tf2-in-python/). Having only used `tf2_ros`, it was a good lead to check out [tf2_geometry_msgs](http://wiki.ros.org/tf2_geometry_msgs).

>**Section 1.3: How to use**
>Please see the Code API documentation (top right of this page) for use.

- Hydro and previous give a page like [this](http://docs.ros.org/hydro/api/tf2_geometry_msgs/html/) (looks like only C++ docs)
- Indigo and following give 404 not founds (example: [Kinetic](http://docs.ros.org/kinetic/api/tf2_geometry_msgs/html/python))

Is this a genuine bug/issue? Perhaps one supposed to just consult the C++ documentation and know it works for python? I'm new, so this is not clear to me. Thanks for any suggestions.

tfoote, 2017-07-10:
That looks like a bug in the documentation generation. I'll have to look into it.

jwhendy, 2017-07-31:
Following back up on this as I'm now learning `moveit` and may have run into the same thing. The link to the API on [the package page](http://wiki.ros.org/moveit_commander) points to [this page](http://docs.ros.org/lunar/api/moveit_commander/html/) which is just empty.

Even clicking around, the docs aren't very helpful. I wanted to know what `moveit_commander.roscpp_initialize()` did. I *think* [this is the page](http://docs.ros.org/lunar/api/moveit_commander/html/namespacemoveit__commander_1_1roscpp__initializer.html#a339c9601770fab25b5dfa5f993c3aacc), but there's no real explanation about it.

Could this be something up with ROS in general (bigger than this package)? Or is `python` just neglected a bit?

tfoote, 2017-07-31:
For moveit_commander there's no content in the mainpage but the docs are generated The classes are documented including the method you are referring to: http://docs.ros.org/lunar/api/moveit_commander/html/namespacemoveit__commander_1_1roscpp__initializer.html  To get there, click on the classes or namespace headers. Though they don't appear to have doc strings so the generated documentation isn't super helpful.

jwhendy, 2017-08-03:
That's the same link I put above. I guess I was optimistically looking for more of a man page. Like what are `args`, what does this do, etc.? I find in a lot of tutorials the "code walkthrough" is pretty vague and is a tautology. E.g. "Here we call `roscpp_initialize` in order to initialize a robot." I guess that's fine, but seeing `roscpp` in a `rospy` tutorial immediately has me wondering why, if it's callings something else behind the scenes that would be helpful to know, etc.

This probably isn't the place to have that discussion, so sorry for contaminating! Initially I just saw the blank docs and thought this was worth adding in case it was a general ros doc issue.

jwhendy, 2018-06-02:
I'm back ~1yr later to confirm this still looks like a big problem. Granted, I'm still noob-ish, but it eats up a lot of my time to try and figure out if migrating code from `tf` to `tf2` should allow me to use something like `tf.TransformListener.transformVector3()`. With literally zero API, and things spread between `tf2`, `tf2_ros`, `tf2_geometry_msgs`, it's surprisingly hard to answer simple questions!

closed 2025-05-31 by sloretz