ros/geometry2 #258: Bug: big memory leak in tf2_ros.Buffer.lookup_transform function

pablo-quilez, 2017-07-25:
# Problem description
The memory use increases very fast after each call to tf2_ros.Buffer.lookup_transform function.
# How to reproduce the error?
1) Start _roscore_ and the following static transform publisher: _rosrun tf static_transform_publisher 1 1 1 0.1 0.1 0.1 /world /ee_link 100_
2) Create a python node containing this slight modified version of the official tf2 tutorial. I just taken out sleeps and topic publishing to increase the number of calls to the lookup_transform in the loop:
```
#!/usr/bin/env python  
import rospy
import math
import tf2_ros
import geometry_msgs.msg
import turtlesim.srv

if __name__ == '__main__':
    rospy.init_node('tf2_turtle_listener')

    tfBuffer = tf2_ros.Buffer()
    listener = tf2_ros.TransformListener(tfBuffer)

    #rospy.wait_for_service('spawn')
    #spawner = rospy.ServiceProxy('spawn', turtlesim.srv.Spawn)
    turtle_name = rospy.get_param('turtle', 'turtle2')
    #spawner(4, 2, 0, turtle_name)

    turtle_vel = rospy.Publisher('%s/cmd_vel' % turtle_name, geometry_msgs.msg.Twist, queue_size=1)

    rate = rospy.Rate(10.0)
    while not rospy.is_shutdown():
        try:
            trans = tfBuffer.lookup_transform('world', 'ee_link', rospy.Time())
        except (tf2_ros.LookupException, tf2_ros.ConnectivityException, tf2_ros.ExtrapolationException):
            rate.sleep()
            continue

        msg = geometry_msgs.msg.Twist()

        msg.angular.z = 4 * math.atan2(trans.transform.translation.y, trans.transform.translation.x)
        msg.linear.x = 0.5 * math.sqrt(trans.transform.translation.x ** 2 + trans.transform.translation.y ** 2)

        #turtle_vel.publish(msg)
```
3) rosrun your_package your_node.py
4) Check with htop or a similar tool the increase on the memory usage.
# Notes
The memory use increases by approx. 15 megabytes per second

pablo-quilez, 2017-07-25:
With tf 1 the memory leak doesn't appear, it is working fine:
```
#!/usr/bin/env python  
import rospy
import tf

if __name__ == '__main__':
    rospy.init_node('tf2_turtle_listener')
    tf = tf.TransformListener()
    while not rospy.is_shutdown():
        if tf.frameExists("/world") and tf.frameExists("/ee_link"):
            t = tf.getLatestCommonTime("/world", "/ee_link")
            position, quaternion = tf.lookupTransform("/world", "/ee_link", t)
        else:
            continue
```
Both produce the same transformation:

tf1:

> (1.0, 1.0, 1.0) (0.04735952982133839, 0.05234912105080037, 0.04735952982133839, 0.9963803086148442)

tf2:

> header: 
>   seq: 0
>   stamp: 
>     secs: 1500982812
>     nsecs: 962030172
>   frame_id: world
> child_frame_id: ee_link
> transform: 
>   translation: 
>     x: 1.0
>     y: 1.0
>     z: 1.0
>   rotation: 
>     x: 0.0473595298213
>     y: 0.0523491210508
>     z: 0.0473595298213
>   ^C  stamp: 
>     secs: 1500982813
>     nsecs:  62109709
>   frame_id: world
> child_frame_id: ee_link
> transform: 
>   translation: 
>     x: 1.0
>     y: 1.0
>     z: 1.0
>   rotation: 
>     x: 0.0473595298213
>     y: 0.0523491210508
>     z: 0.0473595298213
>     w: 0.996380308615

tfoote, 2017-07-25:
Thanks for the reproducable example. I can confirm that I see this from the debian packages. Running it against the development branch `indigo-devel` it looks to be fixed. I suspect #251 resolved the issue and is in the queued release. Is that the same for you?

pablo-quilez, 2017-07-26:
Hello, u are welcome. Thanks to you for your fast answer and nice work! Yes, I think you are right and probably fixing #251 this will be also fix this problem, but to be honest I am not an tf expert. I've just installed it through apt-get. In my computer there is a lunar with tf2 whose package.xml says version 0.5.15. Also this problem happened to me with kinetik (not updated for some months).

tfoote, 2017-07-26:
The fix is released in 0.5.16 which is in the testing repository for lunar if you want to try that out: http://wiki.ros.org/ShadowRepository That will become available in the main repo after the next lunar sync.

0.5.16 is pending for indigo, jade, and kinetic with the fix I've verified so I'm going to close this for now.

closed 2017-07-26 by tfoote