ros/geometry2 #263: interpolate function in cache.cpp outputs wrong time

vsherrod, 2017-08-24:
The interpolate function outputs the timestamp of the first transform instead of the timestamp of the time queried for.

void TimeCache::interpolate(const TransformStorage& one, const TransformStorage& two, TimePoint time, TransformStorage& output)
{
// Check for zero distance case
if( two.stamp_ == one.stamp_ )
{
output = two;
return;
}
//Calculate the ratio
tf2Scalar ratio = double((time - one.stamp_).count()) / double((two.stamp_ - one.stamp_).count());

//Interpolate translation
output.translation_.setInterpolate3(one.translation_, two.translation_, ratio);

//Interpolate rotation
output.rotation_ = slerp( one.rotation_, two.rotation_, ratio);

output.stamp_ = one.stamp_;
output.frame_id_ = one.frame_id_;
output.child_frame_id_ = one.child_frame_id_;
}

It should be:
 output.stamp_ = time;

Instead of 
output.stamp_ = one.stamp_;

tfoote, 2017-08-24:
Confirmed this is a followup to: https://discourse.ros.org/t/interpolate-function-in-cache-cpp-seems-to-output-wrong-timestamp/2489/4

We should both fix this and add a unit test to provide coverage of this output value.

tfoote, 2018-01-19:
Fixed in #265

closed 2018-01-19 by tfoote