ros/geometry2 #270: Bad results from _chainAsVector?

miguelprada, 2017-09-28:
I'm trying to figure out whether I can use `tf2::BufferCore::_chainAsVector` to compute the chain between two frames in a given tree, and I'm having some trouble with it. I have the impression that it does not behave as it should, but I'm not sure I'm using it properly.

Check the very basic test I added in miguelprada/geometry2@ecbf81e2e0a92f6f086607bf1d41aae038e9bbff to see what I'm doing and what I expect the outcome to be.

Am I doing something wrong or is `_chainAsVector` misbehaving?

miguelprada, 2017-09-28:
I'm also trying to understand what is output by the underlying `walkToTopParent` in its `frame_chain` argument to no avail...

See my attempts to understand what's going on in miguelprada/geometry2@255db233f6c8dde32343faf8a02b247f25020fb1 (which require the stubs I added in the previous commit). Here, the result for the _i configuration_ are what I expect, but I cannot wrap my head around what the _v configuration_ outputs. Does it even make sense to use `walkToTopParent` in such a configuration?

Could anyone give me a hint about what should this function **should** output in `frame_chain`?

tfoote, 2017-09-29:
This looks like a bug in the deduplication logic of the forwards and reverse chains. I have a fix locally, but want to write some test for it and remove the debugging code.

miguelprada, 2017-10-05:
I'm in no hurry to fix it, so I'll wait to see what you've got, @tfoote.

In any case, it would be very helpful for me if you can help clarify what the output of these two methods should be. My intuition is that `_chainAsVector` should return the list of frames found when traversing the tree from `source_frame` to `target_frame` while passing through `fixed_frame` (a frame supposedly fixed in time).

I have an even less clear idea about what `walkToTopParent` should return. If it is the chain from a given frame to its top parent, then I don't see why two frame IDs are given as arguments. Or is it rather the chain from one of those frames (`source` or `target`) to their first common ancestor?

tfoote, 2017-10-05:
`walkToTopParent` provides the chain from source to target via the closest shared parent.

`_chainAsVector` provides the chain from source (to fixed if defined) to target. It's basically joining two calls to the above logic at the different times.

miguelprada, 2017-10-06:
Have you, or would you be willing to upload your fix to some branch so I can test it and see whether I can use it?

Also, do you think a PR adding a public stub to `walkToTopParent` would be welcome?

tfoote, 2017-10-16:
Sorry I was testing some more and my fix was only partial. I haven't had time to fully debug the logic issue.

I would rather not expose the walkToTopParent. That's a pretty low level interface with many interconnected elements and potential race conditions. There could be a higher level interface. Part of the value of the tf API is the abstraction that you don't care about the structure of the storage underneath, where walkToTopParent is exposing some of that, but many use cases you'd need to expose much more information. I think that possibly loading the tree out via the YAML interface and then using a python or other scripting language with a library of introspection calls might be more useful as well as more maintainable.

miguelprada, 2017-12-05:
I ended up solving my use-case differently, so I'm no longer interested in this issue. Do you want me to keep it open nevertheless?

tfoote, 2017-12-05:
Yes, I'll keep it open to track the issue and hopefully resolve it.

tfoote, 2018-01-02:
@miguelprada This was a little more involved than I thought. I filled out the tests and fixed the implementation in #278

PR #278 (2018-01-02): Clean up results of _chainAsVector
Simple test to check BufferCore::_chainAsVector.
Unit tests for walk and chain passing now.

Resolves #270
closed 2018-01-02 by tfoote