ros/geometry2 #271: Use of deprecated logging functions

mikepurvis, 2017-10-20:
See: https://github.com/ros/console_bridge/blob/master/include/console_bridge/console.h#L68-L90

https://github.com/ros/geometry2/search?q=logWarn

https://github.com/ros/geometry2/search?q=logError

tfoote, 2017-10-20:
Like: https://github.com/ros/rosconsole_bridge/pull/14#issuecomment-338286124 We can't fix this until we fork development away from Indigo.

mikaelarguedas, 2017-10-20:
I faced the same issue recently on another package. You could work around it if you don't want to branch out just for this change: https://github.com/ros/class_loader/pull/52/files#diff-2777c9ef40eb99c5dbc7f68d8ff4a34f

AlphaJi, 2018-02-27:
I face the same problem, and I have to modify buffer_core.cpp to work around this.

TaylorDixonHD, 2020-03-09:
I faced the same issue, fix was to edit buffer_core.cpp by appending CONSOLE_BRIDGE_ to every occurrence of logError/logWarning (log* in general).

closed 2025-05-31 by sloretz