ros/geometry2 #272: tf2_ros::TransformListener throws on wrong message

AlexisTM, 2017-10-24:
We are using TF2 and AR Alvar tags. Yet the ar_track_alvar occasionally (almost never, yet it happened) sends out a wrong message with NAN. 

```yaml
Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "ar_marker_0" from authority "unknown_publisher" because of a nan value in the transform (nan nan nan) (nan nan nan nan)
         at line 244 in /tmp/binarydeb/ros-kinetic-tf2-0.5.16/src/buffer_core.cpp
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "ar_marker_0" from authority "unknown_publisher" because of an invalid quaternion in the transform (nan nan nan nan)
         at line 257 in /tmp/binarydeb/ros-kinetic-tf2-0.5.16/src/buffer_core.cpp
terminate called after throwing an instance of 'tf2::InvalidArgumentException'
  what():  Quaternion contains a NaN
```

While the message says the transform is ignored, the program actually crashes on a `'tf2::InvalidArgumentException` exception.

AlexisTM, 2017-10-24:
I guess it should be caught here:

https://github.com/ros/geometry2/blob/88abac967f06964a40c94033e8a751cc4c621642/tf2_ros/src/transform_listener.cpp#L115-L128

tfoote, 2017-12-29:
Can you provide a backtrace of the exception? It appears that the denormalized quaternions are being correctly identified and the setTransform is being skipped inside tf2. The exception message appears to be coming from `tf.h` either [here](https://github.com/ros/geometry/blob/indigo-devel/tf/include/tf/tf.h#L398) or [here](https://github.com/ros/geometry/blob/indigo-devel/tf/include/tf/tf.h#L416) Which suggests that it's coming from a completely different part of your program. Probably also effected by the NaN inputs.

AlexisTM, 2018-01-02:
The message is sent from a TF broadcaster (ar_track_alvar):

https://github.com/ros-perception/ar_track_alvar/blob/kinetic-devel/ar_track_alvar/nodes/IndividualMarkers.cpp

I did not have the issue for a while. Probably some parts were still using TF and I switched to TF2 completely in between. 

I will reopen if it happens again.  Thanks @tfoote

closed 2018-01-02 by AlexisTM