ros/geometry2 #273: do_transform_point expects stamped datatypes but ignores header

NikolasE, 2017-11-07:
Related to this question:
https://answers.ros.org/question/275165/attributeerror-point-object-has-no-attribute-point/

do_transform_point expects a PointStamped and a TransformStamped, but does not use the Header-Information of the Point. 

I think the function should also accept a Point (without a Header). 

If however a user passes a PointStamped (with a non-empty frame_id), this frame_id should be compared to the child_frame_id of the Transform and generate at least a warning or even throw an Exception.

tfoote, 2017-12-29:
It is designed for a PointStamped. As per the code comment [`# PointStamped`](https://github.com/ros/geometry2/blob/indigo-devel/tf2_geometry_msgs/src/tf2_geometry_msgs/tf2_geometry_msgs.py#L58)

And yes it should be checking `child_frame_id` matches the `header.frame_id` for full safety/correctness.

A second transform function could be defined to apply to non-stamped points. We've recently added some of those on the c++ side too.

tfoote, 2018-01-02:
Note that there's an inprogress PR #262 which addresses this it just needs some basic unit tests to be merged. A PR to add some basic unit tests for that would make it mergeable. I'm going to close this as a duplicate of that.

closed 2018-01-02 by tfoote