ros/geometry2 #274: tf2_py does not compile for Python 3.5

rochus, 2017-12-05:
The function `static PyObject *_allFramesAsDot` in `tf2_py.cpp` is incompatible with Python 3.5, as it invokes `PyString_FromString`. You can find a (trivial) patch at https://gist.github.com/rochus/aa6d274a349c6aaff692a82fa45af450

astoeckel, 2017-12-06:
Can confirm, just ran into this issue while compiling ROS on Fedora 27 in a Python 3 venv. The linked patch fixes the problem for me.

jacknlliu, 2017-12-20:
Did anyone run into the issue https://github.com/ros/geometry2/issues/259 even with this patch?

tfoote, 2017-12-29:
This same logic appears to be exactly the same as the contents of the method `stringToPython` which you're removing: https://github.com/ros/geometry2/blob/indigo-devel/tf2_py/src/python_compat.h#L17-L24 which is designed to specifically to provide this abstraction between python versions. 
```
inline PyObject *stringToPython(const char *input)
{
#if PY_MAJOR_VERSION >= 3
  return PyUnicode_FromString(input);
#else
  return PyString_FromString(input);
#endif
}
```

tfoote, 2018-01-19:
Since the proposed patch appears redundant and with the help of @allenh1, tf2_py is confirmed to be building with python 3.5 https://gist.github.com/allenh1/3e81e2a6fb28b728dc24416826e99af9 I'm going to close this until there's a reproducible issue.

closed 2018-01-19 by tfoote