ros/geometry2 #285: How to listen to static transform from robot state publisher using the TransformListener?

littlejohnyang, 2018-02-14:
I have set up a robot state publisher with my own robot, and I implemented a listener to obtain the TF message by using TransformListener. I can obtain the /tf information correctly by using the lookupTransform() and allFramesAsString(). However, I can't figure out a way to obtain the /tf_static (as I believe this is a latched topic which doesn't really get published in the same way as /tf. I read the source code and I just couldn't see how I can extract information still if the "message_subscriber_tf_static_" is not even being triggered. 

If using rostopic echo /tf_static, you can definitely see that it has been published but the transform_listener's message_subscriber_tf_static_ has never triggered to call static_subscription_callback(). 

I am also wondering that, instead of using launch file to setup robot_state_publisher, will implementing it in C++ with calling publishFixedTransforms() in a loop solve the problem? (i.e. implementing out own publisher routine for static tf.

tfoote, 2018-02-15:
Please ask questions on http://answers.ros.org according to our support policy: http://wiki.ros.org/Support We try to keep the issue tracker reserved for bugs and feature developments. 

When you do ask there, please provide some code examples and console outputs of what you're trying to do and why it's not doing what you expect it to do.  For example please explain why you think the static transform callback is not being triggered. It's also not clear what the higher level of behavior is that you're looking for and don't get caught in an [XY Problem](https://meta.stackexchange.com/questions/66377/what-is-the-xy-problem)

closed 2018-02-15 by tfoote

littlejohnyang, 2018-02-15:
I will post on answers.ros.org for answer. However, my question is really on whether it actually works with tf_static? If not, it's definitely a bug. The normal tf_echo seems working but not with the tf2 transform listener. 

Also I noticed the subscriber topic inside the TransformListener is hardcoded to "/tf" instead of "tf". Then the ros nodehandle's namespace is not affecting the topic at all. I was hoping it can take different tf topic under different namespace by initialising with the different nodehandle under different namespaces.

littlejohnyang, 2018-02-16:
I updated my question with example code (but they are just pretty much what is provided in the tutorial page.) I want to see a working example for getting static transform (not just the normal tf).