ros/geometry2 #287: tf2_tools echo

lucasw, 2018-02-21:
From https://github.com/ros/geometry/issues/159 @tfoote 

> I would suggest that reimplementing this in python might make sense. Obviously this would be a big change, but it would be a good reason to migrate from tf to tf2_tools where the new python implementation could live. Using argparse and python will make this much simpler to implement.

Use argparse for the extra options (and keep argv[1] [2] for src and target frames for brevity):

--limit=1 only output n many updates then exit

--changed-values-only only update output when transform is more up to date than the last time it was printed out.

Other possible options that occur to me now:

--buffer length

--offset lookup the transform at now() + offset

--time lookup the transform at a specific timestamp.  Maybe the user would set the timestamp in the future, they'd see it generate exceptions for a while, then become available, then roll off the end of the buffer and be unavailable.

The output could be made to be identical to tf_echo.  (Though I'd like an extra digit of precision or two on rotation.)

tf2_echo exists in ros2 https://github.com/ros2/geometry2/blob/ros2/tf2_ros/src/tf2_echo.cpp and looks to be tf_echo.cpp copied with ros2 updates.  Would the desire be to replace that later?

tfoote, 2018-02-21:
Those are all good options to consider adding. And I think shouldn't be too hard if we switch to a python script for command line parsing.

If we switch to python I'd like to do it for both ros and ros2 and deprecate the cpp version of tf2_echo in ROS2. (We had to create tf2_echo for ros2 since geometry won't be ported over so tf_echo isn't available. But we can replace it still.)

lucasw, 2018-02-23:
I'd like to convert the quaternion to euler angles using transformations.py or the transforms3d equivalent, the former would create a tf1 dependency and the latter isn't (yet) in ubuntu or ros deb repositories (#222).  The bare minimum conversion code could be cut and paste into echo.py, or no euler angles for now, or something else?

tfoote, 2018-02-26:
It would be fine to internalize one those conversions from transforms3d for easier user experience. However please make it clearly documented as an internal function. `_` prefix and doc block to say that this is only for internal use and is not a API that can be relied upon.

lucasw, 2018-12-23:
It is possible to `pip install transforms3d` https://pypi.org/project/transforms3d/,  it would be nice to get rid of the borrowed code and use transforms3d proper instead.

tfoote, 2019-01-02:
To use that as a dependency it would be necessary to package it for all native package managers on major supported platforms. (aka debian packages on debian ubuntu, rpms on fedora etc) A pip package cannot be a core dependency.

closed 2025-05-31 by sloretz