ros/geometry2 #293: catkin_make error: buffer_core ‘CONSOLE_BRIDGE_logWarn’ not declared

Acmece, 2018-04-22:
I cloned this repo and used `catkin_make`  to build it, with python2 and python3  (distro:indigo). The errors are all the same as follows:
```
[ 72%] Built target tf2_msgs_generate_messages_cpp
Scanning dependencies of target tf2_msgs_generate_messages
Scanning dependencies of target tf2
[ 72%] Built target tf2_msgs_generate_messages
[ 75%] [ 77%] [ 79%] Building CXX object geometry2/tf2/CMakeFiles/tf2.dir/src/cache.cpp.o
Building CXX object geometry2/tf2/CMakeFiles/tf2.dir/src/buffer_core.cpp.o
Building CXX object geometry2/tf2/CMakeFiles/tf2.dir/src/static_cache.cpp.o
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp: In member function ‘bool tf2::BufferCore::warnFrameId(const char*, const string&) const’:
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:126:49: error: ‘CONSOLE_BRIDGE_logWarn’ was not declared in this scope
     CONSOLE_BRIDGE_logWarn("%s",ss.str().c_str());
                                                 ^
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:134:49: error: ‘CONSOLE_BRIDGE_logWarn’ was not declared in this scope
     CONSOLE_BRIDGE_logWarn("%s",ss.str().c_str());
                                                 ^
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp: In member function ‘bool tf2::BufferCore::setTransform(const TransformStamped&, const string&, bool)’:
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:221:210: error: ‘CONSOLE_BRIDGE_logError’ was not declared in this scope
     CONSOLE_BRIDGE_logError("TF_SELF_TRANSFORM: Ignoring transform from authority \"%s\" with frame_id and child_frame_id  \"%s\" because they are the same",  authority.c_str(), stripped.child_frame_id.c_str());
                                                                                                                                                                                                                  ^
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:227:144: error: ‘CONSOLE_BRIDGE_logError’ was not declared in this scope
     CONSOLE_BRIDGE_logError("TF_NO_CHILD_FRAME_ID: Ignoring transform from authority \"%s\" because child_frame_id not set ", authority.c_str());
                                                                                                                                                ^
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:233:192: error: ‘CONSOLE_BRIDGE_logError’ was not declared in this scope
     CONSOLE_BRIDGE_logError("TF_NO_FRAME_ID: Ignoring transform with child_frame_id \"%s\"  from authority \"%s\" because frame_id not set", stripped.child_frame_id.c_str(), authority.c_str());
                                                                                                                                                                                                ^
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:244:15: error: ‘CONSOLE_BRIDGE_logError’ was not declared in this scope
               );
               ^
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:257:136: error: ‘CONSOLE_BRIDGE_logError’ was not declared in this scope
              stripped.transform.rotation.x, stripped.transform.rotation.y, stripped.transform.rotation.z, stripped.transform.rotation.w);
                                                                                                                                        ^
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:277:279: error: ‘CONSOLE_BRIDGE_logWarn’ was not declared in this scope
       CONSOLE_BRIDGE_logWarn("TF_OLD_DATA ignoring data from the past for frame %s at time %g according to authority %s\nPossible reasons are listed at http://wiki.ros.org/tf/Errors%%20explained", stripped.child_frame_id.c_str(), stripped.header.stamp.toSec(), authority.c_str());
                                                                                                                                                                                                                                                                                       ^
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp: In member function ‘geometry_msgs::TransformStamped tf2::BufferCore::lookupTransform(const string&, const string&, const ros::Time&) const’:
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:636:63: error: ‘CONSOLE_BRIDGE_logError’ was not declared in this scope
       CONSOLE_BRIDGE_logError("Unknown error code: %d", retval);
                                                               ^
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp: In member function ‘void tf2::BufferCore::_chainAsVector(const string&, ros::Time, const string&, ros::Time, const string&, std::vector<std::basic_string<char> >&) const’:
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:1607:63: error: ‘CONSOLE_BRIDGE_logError’ was not declared in this scope
       CONSOLE_BRIDGE_logError("Unknown error code: %d", retval);
                                                               ^
/home/acmece/test/src/geometry2/tf2/src/buffer_core.cpp:1626:63: error: ‘CONSOLE_BRIDGE_logError’ was not declared in this scope
       CONSOLE_BRIDGE_logError("Unknown error code: %d", retval);
                                                               ^
make[2]: *** [geometry2/tf2/CMakeFiles/tf2.dir/src/buffer_core.cpp.o] Error 1
make[1]: *** [geometry2/tf2/CMakeFiles/tf2.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
I can't fix it. Any help will be greatly appreciated.

mikaelarguedas, 2018-04-23:
this looks very similar to https://github.com/ros/geometry2/issues/259#issuecomment-368537754
did you try installing python3-dev ?

Acmece, 2018-04-24:
@mikaelarguedas I did have installed python3-dev.
```
>>> import sys
>>> sys.path
['', '/opt/ros/indigo/lib/python2.7/dist-packages',
 '/home/acmece/catkin_ws/venv/lib/python3.4', 
'/home/acmece/catkin_ws/venv/lib/python3.4/plat-x86_64-linux-gnu', 
'/home/acmece/catkin_ws/venv/lib/python3.4/lib-dynload', 
'/usr/lib/python3.4', 
'/usr/lib/python3.4/plat-x86_64-linux-gnu', 
 '/home/acmece/catkin_ws/venv/lib/python3.4/site-packages']
```
I don't know how to delet this path `'/opt/ros/indigo/lib/python2.7/dist-packages'`
You know, if I didn't source `/opt/ros/indigo/setup.bash`, I can't use `catkin_make` commond.

tfoote, 2018-04-24:
>You know, if I didn't source /opt/ros/indigo/setup.bash, I can't use catkin_make commond.

If you're creating a python 3 environment you cannot use the system installed versions of any of the python tools including `catkin_make` you must use a version from inside your workspace. See http://wiki.ros.org/Installation/Source for an example of how to invoke `catkin_make_isolated` with the path to the version inside your workspace.

closed 2018-04-26 by tfoote

whitecoffee, 2018-06-18:
I get the same problem as you, have u resolved it yet?

tfoote, 2018-06-18:
@whitecoffee you cannot mix python2 and python3. If you need more help please ask a question on answers.ros.org with full instructions how to reproduce your problem.