ros/geometry2 #306: vcstools tar extraction fails for old releases (because repository was renamed?)

gavanderhoorn, 2018-07-05:
Not something directly actionable I believe, but figured I'd report it anyway.

See robust-rosin/robust#63 where we ran into this while investigating a bug in a different repository that needed a checkout of an old Indigo release (`0.5.5`). Repository was called `geometry_experimental` at that time and `rosinstall_generator` appears to prefix that name to the `version` tag of the `.rosinstall` files it generates. This leads to problems when `wstool` tries to extract the downloaded tarball, as it expects a directory that is not there.

Apart from updating the tarballs I'm not sure what would work around/fix this, but that will probably have other consequences I cannot foresee. For our purposes we're manually tweaking the `.rosinstall` files.

tfoote, 2018-07-06:
The tarballs are dynamically generated by github using when you fetch them based on the tags.

Unfortunately I think that this is an unfortunate side effect of github putting the repository name into the tarball filename inside. Which causes the tarball of a tag to change contents if the repo is moved.

[image]

Which basically means that any tarball operations are not robust to repository renaming. The other simple solution would be to update the repo-name in your old/backdated rosdistro file to match the new name, and then the generated content would be right too.

gavanderhoorn, 2018-07-06:
> The tarballs are dynamically generated by github using when you fetch them based on the tags.

Ah, I wasn't aware of that.

> Unfortunately I think that this is an unfortunate side effect of github putting the repository name into the tarball filename inside. Which causes the tarball of a tag to change contents if the repo is moved.

yes, that is what's happening. But the 'problem' only surfaces because `vcstools` is actually checking the contents of the tarball.

>  The other simple solution would be to update the repo-name in your old/backdated rosdistro file to match the new name, and then the generated content would be right too.

that's what we're doing right now.

tfoote, 2018-07-06:
> yes, that is what's happening. But the 'problem' only surfaces because vcstools is actually checking the contents of the tarball.

It's been designed to select a subdirectory and give it a specific name from the tarball. There's potentially an improvement that could be made in vcstool to support some sort of regex or other logic to be more robust. I'm going to close this here as there's nothing actionable here, but it might be worth a discussion on vcstools.

closed 2018-07-06 by tfoote