ros/geometry2 #334: Error in tf2_geometry_msgs.h

rvipin17, 2018-10-09:
`In file included from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:39:0:
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:55: error: ‘array’ is not a member of ‘std’
 geometry_msgs::WrenchStamped toMsg(const tf2::Stamped<std::array<tf2::Vector3, 
                                                       ^
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:55: note: suggested alternative:
In file included from /opt/ros/kinetic/include/ros/serialization.h:45:0,
                 from /opt/ros/kinetic/include/geometry_msgs/Point.h:14,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation.h:35,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation_buffer.h:44,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:37:
/usr/include/boost/array.hpp:60:11: note:   ‘boost::array’
     class array {
           ^
In file included from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:39:0:
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:55: error: ‘array’ is not a member of ‘std’
 geometry_msgs::WrenchStamped toMsg(const tf2::Stamped<std::array<tf2::Vector3, 
                                                       ^
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:55: note: suggested alternative:
In file included from /opt/ros/kinetic/include/ros/serialization.h:45:0,
                 from /opt/ros/kinetic/include/geometry_msgs/Point.h:14,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation.h:35,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation_buffer.h:44,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:37:
/usr/include/boost/array.hpp:60:11: note:   ‘boost::array’
     class array {
           ^
In file included from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:39:0:
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:81: error: spurious ‘>>’, use ‘>’ to terminate a template argument list
 s::WrenchStamped toMsg(const tf2::Stamped<std::array<tf2::Vector3, 2>>& in, geo
                                                                     ^
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:81: error: wrong number of template arguments (2, should be 1)
In file included from /home/addverb/catkin_ws/src/geometry2/tf2_ros/include/tf2_ros/buffer_interface.h:36:0,
                 from /home/addverb/catkin_ws/src/geometry2/tf2_ros/include/tf2_ros/buffer.h:35,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation_buffer.h:45,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:37:
/home/addverb/catkin_ws/src/geometry2/tf2/include/tf2/transform_datatypes.h:44:7: note: provided for ‘template<class T> class tf2::Stamped’
 class Stamped : public T{
       ^
In file included from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:39:0:
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:55: error: ‘array’ is not a member of ‘std’
 geometry_msgs::WrenchStamped toMsg(const tf2::Stamped<std::array<tf2::Vector3, 
                                                       ^
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:55: note: suggested alternative:
In file included from /opt/ros/kinetic/include/ros/serialization.h:45:0,
                 from /opt/ros/kinetic/include/geometry_msgs/Point.h:14,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation.h:35,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation_buffer.h:44,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:37:
/usr/include/boost/array.hpp:60:11: note:   ‘boost::array’
     class array {
           ^
In file included from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:39:0:
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:55: error: ‘array’ is not a member of ‘std’
 geometry_msgs::WrenchStamped toMsg(const tf2::Stamped<std::array<tf2::Vector3, 
                                                       ^
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:55: note: suggested alternative:
In file included from /opt/ros/kinetic/include/ros/serialization.h:45:0,
                 from /opt/ros/kinetic/include/geometry_msgs/Point.h:14,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation.h:35,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation_buffer.h:44,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:37:
/usr/include/boost/array.hpp:60:11: note:   ‘boost::array’
     class array {
           ^
In file included from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:39:0:
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:81: error: spurious ‘>>’, use ‘>’ to terminate a template argument list
 s::WrenchStamped toMsg(const tf2::Stamped<std::array<tf2::Vector3, 2>>& in, geo
                                                                     ^
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1009:81: error: wrong number of template arguments (2, should be 1)
In file included from /home/addverb/catkin_ws/src/geometry2/tf2_ros/include/tf2_ros/buffer_interface.h:36:0,
[... 186 lines cut ...]
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1028:3: note: suggested alternative:
In file included from /opt/ros/kinetic/include/ros/serialization.h:45:0,
                 from /opt/ros/kinetic/include/geometry_msgs/Point.h:14,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation.h:35,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/include/costmap_2d/observation_buffer.h:44,
                 from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:37:
/usr/include/boost/array.hpp:60:11: note:   ‘boost::array’
     class array {
           ^
In file included from /home/addverb/catkin_ws/src/navigation/costmap_2d/src/observation_buffer.cpp:39:0:
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1028:26: error: expected primary-expression before ‘,’ token
   std::array<tf2::Vector3, 2> tmp_array;
                          ^
/home/addverb/catkin_ws/src/geometry2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h:1028:31: error: ‘tmp_array’ was not declared in this scope
   std::array<tf2::Vector3, 2> tmp_array;
                               ^
navigation/costmap_2d/CMakeFiles/costmap_2d.dir/build.make:110: recipe for target 'navigation/costmap_2d/CMakeFiles/costmap_2d.dir/src/observation_buffer.cpp.o' failed
make[2]: *** [navigation/costmap_2d/CMakeFiles/costmap_2d.dir/src/observation_buffer.cpp.o] Error 1
CMakeFiles/Makefile2:36675: recipe for target 'navigation/costmap_2d/CMakeFiles/costmap_2d.dir/all' failed
make[1]: *** [navigation/costmap_2d/CMakeFiles/costmap_2d.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`
I cloned geometry2 into my catkin workspace and after catkin_make, receiving the above error. Any idea how to fix this?

tfoote, 2018-10-11:
Please clarify how to reproduce your workspace including what system you're running on, what packages you have installed from binaries, and where you got them from and what source checkouts you have in your workspace including the branch.

Master-Matt, 2018-11-02:
I just got this error too. I was trying to use navigation from source, but then it required me to use tf2_sensor_msgs from source. I cloned the indigo-devel branch for working on kinetic. I had both of these binary installed before. Do you need to remove them before installing these from source?

tfoote, 2018-11-02:
Yes if you change the target platforms for different branches you may need to update compiler flags. In particular Indigo expects C++03 while Kinetic expects C++11 as the minimum requirement. See REP 3 for more details: http://www.ros.org/reps/rep-0003.html#indigo-igloo-may-2014

Since there's no reply from the original poster and a reproduction of the issue has found a workaround, I'm going to close this.

closed 2018-11-02 by tfoote