ros/geometry2 #341: Multiple TF publishers can clear TF buffer in rospy kinetic

redstorm45, 2018-11-13:
I recently had a problem with TFs clearing with a message similar to "Detected jump back in time of 0.000130s. Clearing TF buffer.", wich caused all of the nodes relying on TF information to break.
Looking at the code, it seems this pull request https://github.com/ros/geometry2/pull/329 was a fault, with the backport in https://github.com/ros/geometry2/pull/336 to Kinetic.

To confirm the issue, I tested with a simple package with multiple TF broadcasters and a single TF listener in python.
The package is available here: https://github.com/redstorm45/poc-tf-kinetic
I was able to reproduce the behavior, but testing on melodic showed no error.
I suspect a change in the way subscribers are handled between the two versions.

I was able to stop the problem from happening by adding a lock to the whole check_for_reset function here: https://github.com/ros/geometry2/blob/764804d8f489730c110e7d28dde682e8d1be3375/tf2_ros/src/tf2_ros/transform_listener.py#L67-L72
Therefore, I think the last_update time is updated while another thread is checking against it, causing a buffer reset (which is not needed in that case).

I don't think this is the right solution, but the bug needs to be addressed in Kinetic.

tfoote, 2018-11-14:
Thanks for the instructions how to reproduce it and debugging into it. The resolution with a lock protecting the now check does suggest there's a race condition. I'm surprised that it's being hit here, but definitely if there's two in flight and it context switches in the middle of this method to another instance it could trigger the reset inappropriately.

Achllle, 2018-11-16:
It looks like apt-get only holds the most recent and broken version of tf2. Is there any way to fix this without installing from source?

tfoote, 2018-11-16:
This effects all versions. There's no older version to roll back to in the binaries. I'll roll a new release shortly. PRs are in for both active development branches. #344 and backport to #345

closed 2018-11-16 by tfoote

carlosjoserg, 2020-12-30:
Hi @tfoote is it possible that the same issue occurs in the C++ side?

I see the `check_for_reset` equivalent here https://github.com/ros/geometry2/blob/noetic-devel/tf2_ros/src/transform_listener.cpp#L106-L111 wihtout the protection applied in https://github.com/ros/geometry2/pull/344

And we are getting the message in https://github.com/ros/geometry2/blob/noetic-devel/tf2_ros/src/transform_listener.cpp#L108 and running out of ideas for the cause.

Thanks

tfoote, 2021-02-11:
I guess there's no reason it shouldn't be able to happen in c++ too. If you're getting that error that's about the only option as long as you believe time is monotonic. Have you validated that you're not actually getting time jumps possibly from something like an NTP client?

It's a very short window and would have to be a race likely with thread preemption. If you can add a lock there to protect and that resolves it then it's more of an issue than I'd have imagined.

carlosjoserg, 2021-02-11:
Hi @tfoote 

We actually have an update on it, and yes, you are right, it is a problem with both, not having monotonic time in some parts and also NTP corrections, in our case.

While investigating, we implemented the same python [test mentioned above](https://github.com/ros/geometry2/issues/341#issue-380396039) in [C++](https://github.com/carlosjoserg/poc-tf-kinetic) and in fact we didn't experience the issue without any lock-protection in the C++ side, fortunately.

For the monotonic part, we replicated the current `ros::Rate` using a SteadyClock, [like this](https://gist.github.com/carlosjoserg/5e29aac245bc5cbb364200ae621cb0ed).

All together, we are good so far. Many thanks again for the support, and confirming our findings. You can keep the issue closed ;)

PR #345 (2018-11-16): Protect the time reset logic from a race condition.
Fixes #341

This could incorrectly trigger a buffer clear if two concurrent callbacks were invoked.
commit 32404a3a (2018-11-16): Protect the time reset logic from a race condition.
Fixes #341

This could incorrectly trigger a buffer clear if two concurrent callbacks were invoked.
PR #344 (2018-11-16): Protect the time reset logic from a race condition.
Fixes #341

This could incorrectly trigger a buffer clear if two concurrent callbacks were invoked.
commit 2d546523 (2018-11-16): Protect the time reset logic from a race condition.
Fixes #341

This could incorrectly trigger a buffer clear if two concurrent callbacks were invoked.