ros/geometry2 #342: No package 'bullet' found

kaixin-bai, 2018-11-14:
ubuntu16.04 
ROS-Kinetic 
branch: indigo-devel

wenn catkin_make, i got the Error below.

-- +++ processing catkin package: 'tf2_bullet'
-- ==> add_subdirectory(geometry2/tf2_bullet)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'bullet'
--   No package 'bullet' found
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  geometry2/tf2_bullet/CMakeLists.txt:7 (pkg_check_modules)

-- Configuring incomplete, errors occurred!
See also "/home//Desktop/11/build/CMakeFiles/CMakeOutput.log".
See also "/home//Desktop/11/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

with ‘sudo apt-get install ros-kinetic-tf2-bullet’ the problem solved.

kaixin-bai, 2018-11-14:
> ubuntu16.04
> ROS-Kinetic
> branch: indigo-devel
>
> wenn catkin_make, i got the Error below.
>
> -- +++ processing catkin package: 'tf2_bullet'
> -- ==> add_subdirectory(geometry2/tf2_bullet)
> -- Found [...]

closed 2018-11-14 by kaixin-bai

kennydedavion, 2022-01-27:
Hi, Im same problem. 
`==> Processing catkin package: 'tf2_bullet'
==> Building with env: '/opt/ros/kinetic/env.sh'
==> cmake /home/pi/ros_catkin_ws/src/geometry2/tf2_bullet -DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel_isolated/tf2_bullet -DCMAKE_INSTALL_PREFIX=/opt/ros/kinetic -DCMAKE_BUILD_TYPE=Release -G Unix Makefiles in '/home/pi/ros_catkin_ws/build_isolated/tf2_bullet'
-- Checking for module 'bullet'
--   No package 'bullet' found
CMake Error at CMakeLists.txt:11 (find_package):
  By not providing "Findbullet.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "bullet", but
  CMake did not find one.

  Could not find a package configuration file provided by "bullet" with any
  of the following names:

    bulletConfig.cmake
    bullet-config.cmake

  Add the installation prefix of "bullet" to CMAKE_PREFIX_PATH or set
  "bullet_DIR" to a directory containing one of the above files.  If "bullet"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/pi/ros_catkin_ws/build_isolated/tf2_bullet/CMakeFiles/CMakeOutput.log".
<== Failed to process package 'tf2_bullet': 
  Command '['/opt/ros/kinetic/env.sh', 'cmake', '/home/pi/ros_catkin_ws/src/geometry2/tf2_bullet', '-DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel_isolated/tf2_bullet', '-DCMAKE_INSTALL_PREFIX=/opt/ros/kinetic', '-DCMAKE_BUILD_TYPE=Release', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/pi/ros_catkin_ws/build_isolated/tf2_bullet && /opt/ros/kinetic/env.sh cmake /home/pi/ros_catkin_ws/src/geometry2/tf2_bullet -DCATKIN_DEVEL_PREFIX=/home/pi/ros_catkin_ws/devel_isolated/tf2_bullet -DCMAKE_INSTALL_PREFIX=/opt/ros/kinetic -DCMAKE_BUILD_TYPE=Release -G 'Unix Makefiles'

Command failed, exiting.
pi@raspberrypi:~/ros_catkin_ws :  sudo apt-get install ros-kinetic-tf2-bullet
Načítají se seznamy balíků… Hotovo
Vytváří se strom závislostí       
Načítají se stavové informace… Hotovo
E: Nelze najít balík ros-kinetic-tf2-bullet 
`
command `sudo apt-get install ros-kinetic-tf2-bullet` not solve.

BradfordMedeiros, 2022-05-24:
So I'm approaching this from a different project (not ros), but the same issue coming up which seems to be a warning from FindBullet.  As a result of not finding the 

[image]

However, when I actually look in my source files I notice this: 
[image]

I'm not sure why this isn't being built, but if you simply copy  it and rename it bulletConfig.cmake  it starts to build.

I don't know the effect of simply renaming it, I don't know if the lower case name is supposed to be generated somehow, etc, simply that doing this makes it build. 

Perhaps findBullet needs to be updated, or it's something to do with my version of Cmake or something, I don't have that figured out.  I'm also not sure if there's problems with the simple copy/paste solution, but it -- seems to work --