ros/geometry2 #349: Unable to locate package ros-kinetic-tf2-tools

aPonza, 2018-12-06:
In [Introduction to tf2](http://wiki.ros.org/tf2/Tutorials/Introduction%20to%20tf2) users have to install `ros-$ROS_DISTRO-tf2-tools` which however doesn't exist. This addition comes from [this answer](https://answers.ros.org/question/263285/rosrun-tf2_tools-not-found/) most likely.

I can instead find the package `tf2-tools - Robot OS tool for tf2 transform library second generation` in aptitude. This is most likely what was meant, but before updating the wiki with the correct name I wanted confirmation that something happened and the package was renamed (confirmation which I can't find).

Is this the case?

aPonza, 2018-12-06:
That is not the case, as after installing also said `tf2-tools` package, once at the appropriate point in the tutorial, I get
```
$ rosrun tf2_tools view_frames.py
[rospack] Error: package 'tf2_tools' not found
 ```

aPonza, 2018-12-06:
I afterwards saw there's no kinetic-devel branch, I figured that might be the reason for the missing package.

Regardless I downloaded the folder (from melodic-devel) and everything seems to work.

If everything is as intended this can be closed.

tfoote, 2018-12-06:
You can manually install `ros-kinetic-tf2-tools` from apt. The referenced tutorial is missing it as a dependency that will automatically install it. 

Note that although the melodic-devel branch generally is compatible, the kinetic development branch is still indigo-devel as can be seen in the rosdistro or here: https://wiki.ros.org/tf2_tools?distro=kinetic

closed 2018-12-06 by tfoote