ros/geometry2 #356: Unlimited buffer caching time

christian-rauch, 2019-01-24:
I am using a `tf2_ros.Buffer` to collect transformations via `set_transform` to later synchronize the export of messages into other formats (images, csv, ...).
By default, the Buffer has a cache time of 10 seconds. For scenarios where the Buffer is used in an offline application, it would be useful to have an option to cache all transformations in the buffer.

The current workaround would be to just chose a very large caching time. It would be better if this could be made more explicit when constructing the buffer, e.g. by setting `cache_time=genpy.Time(0)` or something like `cache_all=True`.

tfoote, 2019-01-25:
Unlimited caching in itself is moderately dangerous for memory usage. I agree that a flag like that would be clearer when using the API. Adding the extra complexity to the system isn't very attractive when you could just pass an argument that is larger than any possible runtime for the system. I don't think that we have a Duration::Max implementation but if you passed in something like as a constant it would be pretty clear reading the code, and the behavior would not be any different.

christian-rauch, 2019-01-25:
Caching transforms indefinitely is of course dangerous. But for an offline use case, where a user decides to cache all (finite amount of) transforms from a source (e.g. bagfile), an option to explicitly tell the buffer to cache everything would improve the code quality.

At the moment, I am defining the buffer like
```Python
tf_buffer = tf2_ros.Buffer(cache_time=genpy.Time(secs=pow(2,31)-1))
```
i.e. with the largest possible time value. However, this is not straight forward, since you first need to determine the largest possible time value by trial and error.
A `genpy.Time.max()` would indeed help to make it explicit that 2^31-1 is not an arbitrary value, but the largest possible time span ROS can represent.